betaflight/betaflight-configurator #2608: Bug on PID-Slider : calculation differ on same slider values (MacOS)

mrRobot62, 2021-09-19:
Used FW & Configurator : same as in #2605 
[image]

Testszenario
1) go to PID enable slider, set all sliders to 1 and get this values
[image]
2) set DGain slider to 2 - get this values
[image]
3) change back to 1 and get values as seen in (1)
4) change back to 2 and this values
[image]
5) if I do this several times I get several calculations
6) this bug is **not always reproducible** sometimes it works as supposed, sometimes not. I did this check only on DGain slider

haslinghuis, 2021-09-19:
Seems you don't have the correct firmware. We still have to change some defaults in the firmware. PID values with slider position in middle should look like:
[image]

mrRobot62, 2021-09-19:
???? - that‘s obfuscated :-O
For saftey reason, could you resend link to FW which I should use :-O 
mhm, maybe I rearranged values during my test - not sure - but could you see the described effekt on you test environment?

mhmmmmmm

haslinghuis, 2021-09-19:
FW is here but the new defaults have to be added yet: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=3887&view=artifacts&pathAsName=false&type=publishedArtifacts

Also configurator has been updated. So you might need to grap the latest version again at: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=3917&view=artifacts&pathAsName=false&type=publishedArtifacts

mrRobot62, 2021-09-19:
Hi,
installed latest Betaflight FW as suggested (betaflight_4.3.0_STM32F405_55a67977f.hex)
installed lates Betaflight-Configurator as suggested

Bug?
- "Auto detect" deactivated if no board is selected. I assume it should be activated if NO board is detected and deactivated if a board is selected

Betaflight-Configurator (Build)
[image]

PID-Slider test (Default) (Expert-Mode OFF)
[image]

Set Damping to 2.0
[image]

Play around moving Damping Slider
- no calculation error. Bug fixed

Flash FW again to avoid any misconfiguration in memory
Start PID Mask, **Expert-Mode ON**
[image]

Move Damping Slider to 2.0
After several movings the slider from left to right
[image]

[image]

Bug not fixed

mrRobot62, 2021-09-19:
# Update 1
## Moving Tracking (P&I) left/right several times
- same calculation problem as with Damping-Slider

## Moving Stick-Response
- ok - no problem same together with Master-Slider

## Dynamic-Damping (DMax)
- same problem as with Damping DGain Slider 
### Value set to 2.0
[image]

[image]

## Drift-Wobble
- same problem as Damping

## Pitch-Damping
- same problem as Damping

## Pitch-Tracking
- same problem as Damping

[image]

[image]

# Conclusion
While testing the sliders, I noticed that the error rate depends on the speed at which the user moves the slider.
The faster he moves, the higher the probability that the calculation is incorrect.
The slower the user moves, the higher the probability that the result is correct.

haslinghuis, 2021-09-19:
@mrRobot62 thanks for testing, I can reproduce. Now added the mouseup next to the input element so it recalculates after releasing the mouse.
Please test again with new configurator (please wait for the build has done).

I like the issue to be in one place as they are related and otherwise duplicate.

mrRobot62, 2021-09-19:
ok, will do further testings tomorrow, should I flash new firmware too?

haslinghuis, 2021-09-19:
Thanks @mrRobot62 for testing. Closing now.

FYI: I've committed the needed defaults in firmware too.

closed 2021-09-19 by haslinghuis

mrRobot62, 2021-09-19:
perfect :-) - good job @haslinghuis