betaflight/betaflight-configurator #2609: Improvement or Bug? - Auto-Detect deactivated if no board selected

mrRobot62, 2021-09-20:
Not a bug but nearby ;-)
On update Firmware page the *Auto detect* Button is a great idea, but this button should enabled if NO board is selected. Currently user have to select something, than button is activated and user can use it.

That's little bit confusing.

In my opinion : Low priority

Latest BF-Configurator Build from yesterday : 
[image]

[image]

[image]

haslinghuis, 2021-09-20:
This redefined function should fix it:
```
        function updateDetectBoardButton() {
            const isDfu = portPickerElement.val().includes('DFU');
            const isBusy = GUI.connect_lock;
            const isLoaded = self.releases ? Object.keys(self.releases).length > 1 : false;
            const isAvailable = PortHandler.port_available || false;
            const isButtonDisabled = isDfu || isBusy || !isLoaded || !isAvailable;

            detectBoardElement.toggleClass('disabled', isButtonDisabled);
        }
```

mrRobot62, 2021-09-22:
Hi @haslinghuis made the chang on my local machine and tested it locally. Not sure what are the next steps :-O. I forked this repository. Should I checkin only - or should I do some more steps. First time for me in kind of a virtual development team on git.

haslinghuis, 2021-09-22:
Above fix can be tested with the artifact from the PR mentioned above: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=3934&view=artifacts&pathAsName=false&type=publishedArtifacts

mrRobot62, 2021-09-22:
Oh jesus - I totally misunderstood you comment above :-O - haha - yeah know I installed bf-configurator sources and can play around it :-)

Seems this issue is fixed

PR #2554 (2021-09-26): Remove scrolling conditional and allow auto-detect board when no board selected
Fixes: #2551
Fixes: #2609
closed 2021-09-22 by mrRobot62