betaflight/betaflight-configurator #2610: Option "Permanently activate expert mode" works only if BF-Configurator is restarted

mrRobot62, 2021-09-20:
New function on "Option "Permanently activate expert mode" Tab did not work perfect.

# BF-Configurator Version
[image]

[image]

# Problem
[image]
If switch is ON and user connect to FC on all tabs with Expert-Mode switch, this switch is activate - perfect.
If user switch on a mask (e.g. PID Tuning) switch OFF and disconnect from FC and reconnect again. Expert-Mode on PID-Tuning is OFF.

# Conclusion
Suggestion:  always read Option-states during connect not on start. This increase user experience

mrRobot62, 2021-09-20:
mhm, but if expert-mode is set to on as base option, than my gut feeling is, users expectation is: Expert mode is ON as long I change it to off during a session and it’s ON again if disconnect/connect again. Because it’s a system wide configuration. Otherwise I do not understand why we need this option

As an other example: if user set in base options the flag "reopen last tab". if this switch is on, bf-conf store the latest tab and after restarting or re-connect, system jumps to users latest tab. Thats the expectation.

Same analogy should be done with "always expert mode on". If user set this flag at every time expert mode is on, even he/she restart the app or disconnect/connect.

haslinghuis, 2021-09-20:
Have to check later as my drive has no space left. Please check the other issue.

haslinghuis, 2021-10-21:
@mrRobot62 happy to assist you to make this happen if you like to contribute to the project.

sobelek, 2022-03-20:
@haslinghuis 
Hey, if this issue is still open and up for grab I would like to give it a try

haslinghuis, 2022-03-20:
@sobelek
Sure take a hit on it. Much appreciated.

PR #2866 (2022-03-29): Make expert mode permanent even after manually switching it off and reconnecting
Hey.
This PR Fixes #2610
If you see any other possibility of implementing this please let me know and lets talk about it.

This is my first commit to this codebase so if I missed anything process related let me know.
closed 2022-03-29 by blckmn