betaflight/betaflight-configurator #2629: Installation taking forever with 10.7.1

pdxtigre, 2021-10-11:
Latest betaflight-configurator-installer_10.7.1_win32.exe taking forever to install.
It is searching the files in node_modules and hogging CPU at 18%.
Can anything be done to speed this process up?

[image]

Nikotttin, 2021-10-11:
[image]

seems like it is calling somewhere without response

aka13-404, 2021-10-16:
I don't get it. how the listing is so slow and CPU intensive at the same time. I have it installed on an m2 ssd, and it is taking for ever.
I'd go with a zipped version, but sadly there is none, only the installer. 

Should be definately a warning in the release notes, killed my plans quickly tweaking some settings.

grabejud, 2021-10-17:
[image]
too late to read this. got the bug. uninstallaed and installed. cleaned registry, deleted appdata files, and this thing is stuck

ggmaniack, 2021-10-17:
@grabejud 
It will finish installing, it just takes ages.

haslinghuis, 2021-10-18:
As suggested by @McGiverGim uninstall the current version before installing this one.

closed 2021-10-18 by haslinghuis

haslinghuis, 2021-10-18:
Added a note in the release to uninstall previous version

Asizon, 2021-10-19:
Why you dont report this issue in 10.7.0??it is the same behaviour.

pdxtigre, 2021-10-19:
> Added a note in the release to uninstall previous version

Thanks a lot!  That's what is needed.  

> Why you dont report this issue in 10.7.0??it is the same behaviour.

Asking why is not helpful without understanding the background.  I am not a fan of installer in general, I prefer the zip package.  I do not install BF on a daily basis and only install when I need.  I happened to run into this specific version to start with and realized the problem, thus, I wanted to report so others can be aware of what is going on.  Not to mention, the latest available releases are listed here [betaflight-configurator/releases](https://github.com/betaflight/betaflight-configurator/releases), do I see a big disclaimer?  Nope.  Saving people time from running into the problem is much appreciated.

ThisIsMyNameBecauseEverythingIsTaken, 2021-10-31:
> As suggested by @McGiverGim uninstall the current version before installing this one.

Incorrect.
Brand new windows install. BF configurator has never been installed on this computer. Still hangs on installation.

mrcottonmouth, 2021-11-02:
Same issue 10.7.1 on Windows 11

mwhidden, 2021-11-10:
> 
> 
> > As suggested by @McGiverGim uninstall the current version before installing this one.
> 
> Incorrect. Brand new windows install. BF configurator has never been installed on this computer. Still hangs on installation.

I second @ThisIsMyNameBecauseEverythingIsTaken : I tried installing on a new machine, no prior installation of Configurator, and it hung in exactly the spot described above.

rickyyR, 2021-11-13:
Same problem on multiple windows machines, creating the uninstall taking forever at @fortawesome fonts even with completely clean install. 10.7.0 is installing fine but should not be used apparently...

randstorm, 2021-11-15:
Tried to tweak some stuff quickly. Found out after uninstalling bf on two seperate computers and reinstalling bf 10.7.1 on both and hung up at the end on both. So letting it take it's course???? It's all I can do. This is very much time wasting at a very inconvenient time for me. Is it gonna finish? Anybody install 10.7.1 successfully?

rickyyR, 2021-11-16:
> Tried to tweak some stuff quickly. Found out after uninstalling bf on two seperate computers and reinstalling bf 10.7.1 on both and hung up at the end on both. So letting it take it's course???? [...]

Yes it willl work, just take some time. Tbh for me personaly it was going faster after installing 10.7.0 first, could have been luck tho.

haslinghuis, 2021-11-19:
Okay, this is problem now. It's takes a very long time. However as this is fixed already in 10.8 we are working to get this released asap.

wernerterreb, 2021-12-20:
I wonder how long....   mine has been sitting in the same state shown below for about two hours so far.   I will leave it to wait longer, but I am not very hopefull.

[image]

AngryPepper, 2022-01-03:
How long will it take to download BetaFlight

wernerterreb, 2022-01-09:
Just for the record... it does eventually install.  Just takes very long.  And as @haslinghuis mentioned in an earlier comment... the issue will be fixed in a future release.  I think its just 10.7 that has this problem.

Asizon, 2022-01-09:
@wernerterreb finally fixed on released 10.7.2