betaflight/betaflight-configurator #2631: OSD scaling for larger pc monitors

UglyBabyGaming, 2021-10-11:
I have a large 4k 42inch monitor but the OSD layout is around the size of a phone making it difficult to place objects.
the Speedy Bee app shows this better but is harder to place if you have fat fingers.

please could you implement a zoom/scale function to solve this so it works on different size monitors so we can all see what we are doing.

even placement of the OSD preview top or bottom of the page so it stays large and easier to use.

McGiverGim, 2021-10-11:
I tried to do something some time ago with css but I didn't reach to something usable. Maybe other with more time can try it.
With monitors over 1080p is clear that we need something...

UglyBabyGaming, 2021-10-11:
thanks for getting back so quickly,
can you point me in the right direction to where to start looking for how the Beta flight overall layout is produced..
it seem to me to be the same solution as webpage scaling depending on how it is produced in Beta flight

closed 2022-05-20 by haslinghuis