betaflight/betaflight-configurator #2649: Can´t install on Windows 10 x64

MtsPac, 2021-10-30:
I tryed to install the 10.7.1 version and it stops at the end of the installation. I've stopped the process and tried to open the betaflight configurator and it worked, but the version was the 10.4.1. Don,t know whats wrong. Tried to unninstall, erase the registration, rebooting. Nothing seems to work. Everytime it stops at the same point and installs the 10.4.1 version.

haslinghuis, 2021-10-30:
Duplicate of #2629

Fixed for 10.8.0

Polax83, 2021-11-14:
Hello, same problems.
When did 10.8.0 release?

rwhazlehurst, 2021-11-19:
Same problem...see screenshot for point where it hangs.
[image]
.

rwhazlehurst, 2021-11-19:
> Same problem...see screenshot for point where it hangs. [image] .

Ugg...not 2 minutes after pressing submit on my previous post, I jumped over to the installer to kill the job and it had completed successfully...

butcher51, 2021-11-21:
Yeah same here, installer got stuck.

cremymeatballs, 2021-11-21:
Ive tried several times myself. Mine gets stuck in the exact same spot every time I've let it run for almost two days now and it still hasn't installed what could be the problem?

ConnerC579, 2021-12-26:
[image]
I am having this problem. Have tried rebooting and restarting install process.

haslinghuis, 2021-12-26:
We have a 10.8 pre-release. Please check it out: https://github.com/betaflight/betaflight-configurator/releases

Please be aware to update any 4.3 development releases with the latest available firmware.

ConnerC579, 2021-12-26:
Thank you for the response. I was able to install 10.8. I don’t have any
receivers or drones to use yet. But I will soon.

Can you point me to some good beginner resources to learn how to use
betaflight? Right now it’s a foreign language to me. All of it.

haslinghuis, 2021-12-26:
https://github.com/betaflight/betaflight/wiki

Bud13420, 2021-12-29:
> [image] I am having this problem. Have tried rebooting and restarting install process.

Having same issue? How fixed?

ConnerC579, 2021-12-29:
Try this 10.8 version. It installed for me when I had this issue.

https://github.com/betaflight/betaflight-configurator/releases

Bud13420, 2022-01-01:
I did. Thanks brother/sister!

closed 2022-01-19 by haslinghuis