betaflight/betaflight-configurator #2652: Node Version 10 reached end of life on April 30th 2021

wind0r, 2021-11-08:
Currently .nvmrc, travis and Jenkins are using Node 10 which is end of life.

Jenkins Container Logs:
```
================================================================================
================================================================================

                              DEPRECATION WARNING

  Node.js 10.x is no longer actively supported!

  You will not receive security or critical stability updates for this version.

  You should migrate to a supported version of Node.js as soon as possible.
  Use the installation script that corresponds to the version of Node.js you
  wish to install. e.g.

   * https://deb.nodesource.com/setup_12.x — Node.js 12 LTS "Erbium"
   * https://deb.nodesource.com/setup_14.x — Node.js 14 LTS "Fermium" (recommended)
   * https://deb.nodesource.com/setup_16.x — Node.js 16 "Gallium"

  Please see https://github.com/nodejs/Release for details about which
  version may be appropriate for you.

  The NodeSource Node.js distributions repository contains
  information both about supported versions of Node.js and supported Linux
  distributions. To learn more about usage, see the repository:
    https://github.com/nodesource/distributions

================================================================================
================================================================================

```

McGiverGim, 2021-11-08:
There is a work in progress... https://github.com/betaflight/betaflight-configurator/pull/2367

wind0r, 2021-11-08:
Ohh. Thanks! 
I somehow didn't found it :) 

Thanks and Closing :)

closed 2021-12-20 by ctzsnooze

wind0r, 2021-12-22:
Jenkins also has the node update now :)