betaflight/betaflight-configurator #2654: Crashed on MacOS

COOLFPVkevin, 2021-11-10:
I am using a 2021 MacBook Pro with M1 chip. I connected my fpvdrone with an adapter to the laptop. When I open the Betaflight and after I press the connect, after few seconds it will crashed. My software is macOS Monterey 12.0.1

Six6siX, 2021-11-14:
Same random crashes on Mac Mini M1. It lost the normal graphics depicting the quad. See screenshot as an example.

[image]
[image]

haslinghuis, 2021-12-15:
Please try a nightly build

k4m454k, 2021-12-20:
same problem on m1

haslinghuis, 2021-12-20:
Please try:

https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=4996&view=artifacts&pathAsName=false&type=publishedArtifacts

haslinghuis, 2021-12-20:
And maybe also 

https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=4730&view=artifacts&pathAsName=false&type=publishedArtifacts

Asizon, 2021-12-28:
@k4m454k did you try attached development builds??

niceric3, 2022-01-06:
Running the following command resolved the issue for me: `codesign --remove-signature /Applications/Betaflight\ Configurator.app/Contents/Frameworks/nwjs\ Framework.framework/Helpers/nwjs\ Helper\ (Renderer).app`

k4m454k, 2022-01-06:
> @k4m454k did you try attached development builds??

yep. works fine

Asizon, 2022-01-07:
Already fixed

closed 2022-01-07 by Asizon