betaflight/betaflight-configurator #2662: Cant use Firmewareupdate

Spookydo, 2021-11-21:
Hi,

I want click on update Firmeware and Betaflight is still loading forever.
[https://www.youtube.com/watch?v=t82k51ppdzs](url)
at DevTools i find this.

[https://prnt.sc/20cg2qx](url)

ctzsnooze, 2021-11-22:
The video shows that the flight controller is not in DFU mode.

Normally - or for many boards, at least, including most of mine - the board must be in DFU mode before clicking Update Firmware.

limonspb, 2021-11-22:
Looking through the video... This is weird. Windows showing ports like Linux-Mac style?
What system is that?
[image]

haslinghuis, 2021-11-22:
@Spookydo - you could test with a nightly build. Never saw an issue with manifest on windows.

closed 2021-12-26 by Asizon