betaflight/betaflight-configurator #2663: 18年MacBook pro上下载了Betaflight 10. 7.1，但是端口不供电，飞控指示灯不亮

123Ligyy, 2021-11-21:
18年的苹果MacBook pro下载了Betaflight 10.7.1版本，连上飞控不供电，换数据线也无法解决，请问是什么问题

DusKing1, 2021-11-21:
English please

haslinghuis, 2021-11-21:
请问这里可以用英文吗？(Can you please use English here?)

Question: The Apple MacBook Pro of 18 years has downloaded Betaflight version 10.7.1. The flight controller does not have power supply, and it cannot be solved by changing the data cable. What is the problem?

如果您使用的是 18 年的 Apple MacBook Pro，那么它就不再适用了 (If you are using an 18-year-old Apple MacBook Pro, then it is no longer applicable) :joy: 

您可以自行承担风险测试 Betaflight 10.8 版每晚构建。(You can test the nightly Betaflight 10.8 build at your own risk)

如果 FC 没有显示任何 LED，则可能处于 DFU 模式或 FC 的 USB 端口不再工作。(If the FC does not display any LEDs, it may be in DFU mode or the FC's USB port may no longer work)

如果您可以在将智能手机连接到 Mac 的情况下通过同一根电缆传输文件，则该电缆应该没问题。(If you can transfer files over the same cable with your smartphone connected to the Mac the cable should be ok)

haslinghuis, 2021-11-21:
# 如果您的问题看起来像是硬件故障或配置问题，请不要在这里提出问题。

## 请考虑使用其他用户支持选项，例如询问您使用的硬件的制造商，RCGroups：https://rcgroups.com/forums/showthread.php?t=2464844，或 Slack（在 https:// slack.betaflight.com/) 或其他用户支持论坛和群组（例如 Facebook）。

## 如果您认为固件本身存在问题，请按照以下步骤操作：
1. 描述您的问题；
2. 包括重现问题的方法；
3. 提供尽可能多的关于您的硬件和软件的信息，包括：
- 您使用的硬件/软件；
- 使用的所有软件的版本；
- 事物是如何连接/接线的。
4. 删除此文本 :)。

closed 2021-11-21 by haslinghuis