betaflight/betaflight-configurator #2664: Android APP: Can't save blackbox file - console error

limonspb, 2021-11-24:
Current Master 10.8.
Last commit:
```
commit dd183f041200b2b095f9a3eefe0a8fbff786acbc (HEAD -> master, upstream/master)
Merge: 41747c65 4c9fbc6a
Author: J Blackman <blckmn@users.noreply.github.com>
Date:   Tue Nov 9 18:17:18 2021 +1100

    Auto merged - #2634 at Tue, 09 Nov 2021 07:17:18 GMT

    Split header and fix i18n message
```

Can't save a BlackBox file from the Android app - probably a permission issue, but the app did not ask me about anything after installation. Same problem occurs when saving a backup from presets tab
Console says the following:

```
Chrome API Error: FILESYSTEM (adapted from Cordova): [object Object].
 Traced Error
    at checkChromeRuntimeError (http://localhost/js/utils/common.js:1537:90)
    at http://localhost/js/tabs/onboard_logging.js:612:17
    at chromeCallbackWithError (http://localhost/js/cordova_chromeapi.js:20:18)
    at http://localhost/js/cordova_chromeapi.js:341:25
    at fail (http://localhost/plugins/cordova-plugin-file/www/DirectoryEntry.js:77:9)
    at Object.callbackFromNative (http://localhost/cordova.js:289:52)
    at <anonymous>:1:9
checkChromeRuntimeError @ common.js:1537

```
[image]

[image]

limonspb, 2021-11-24:
@WalcoFPV any ideas? Thanks!

Asizon, 2021-11-24:
Yes, this save file system is outdated, i think apk is ready to put your phone in masive storage to then save your logs.

limonspb, 2021-11-24:
> Yes, this save file system is outdated, i think apk is ready to put the flight controller in masive storage to then save your logs.

@Asizon 
So there is no way to save ANYTHING to a local file? Because i am having the same problem with saving txt file. (presets tab, save backup config)

git bisect showed that the issue started happening after
https://github.com/betaflight/betaflight-configurator/issues/2664
by @mikeller

WalcoFPV, 2021-11-24:
Looks like, as @Asizon said, the save file system is outdated. With #2577, the `android-targetSdkVersion` was set to 30 and before we juste defined `android-minSdkVersion` to 19. So it's possible that the SDK version used has changed. The cordova plugin used for filesystem could not be compatible with this new SDK version

limonspb, 2021-11-24:
> Looks like, as @Asizon said, the save file system is outdated. With #2577, the `android-targetSdkVersion` was set to 30 and before we juste defined `android-minSdkVersion` to 19. So it's possible [...]

I am a little confused now. Is there a way to save any (TXT) file into Android phone internal storage now from the Configurator?
I am using
`chrome.fileSystem.chooseEntry`
works good on Desktop, just not on Android anymore

WalcoFPV, 2021-11-24:
`chrome.fileSystem` doesn't exist on Android. I had to defined it in `src/js/cordova_chromeapi.js` using Cordova FileSystem

haslinghuis, 2021-11-24:
It only works with external memory (SD Card) or maybe cloud storage.

haslinghuis, 2021-11-24:
@limonspb Can you test saving on external storage (SD Card)?

limonspb, 2021-11-24:
> @limonspb Can you test saving on external storage (SD Card)?

Not sure how this is done. The path for saving is picked automatically

WalcoFPV, 2021-11-25:
The issue seems to be related to [the new Android 11 storage rules](https://developer.android.com/about/versions/11/privacy/storage). Apps can no longer create their own directory in the root directory of the device. `/Android/data/betaflight-configurator` is still writable : instead of saving files in `/Betaflight configurator.`, we can save files in it as a workaround

limonspb, 2021-11-25:
> The issue seems to be related to [the new Android 11 storage rules](https://developer.android.com/about/versions/11/privacy/storage). Apps can no longer create their own directory in the root [...]

Interesting. Why I was able to still use old version of the app to save files on the roof directory/betaflight?

WalcoFPV, 2021-11-25:
@limonspb Because of the `targetSdkVersion` set to 30 in #2577. The app is now targetting devices with Android 11. Before, the `targetSdkVersion` was lower and the new Android 11 rules were not applied if I understand correctly

limonspb, 2021-11-27:
Closed with this fix by @WalcoFPV:
https://github.com/betaflight/betaflight-configurator/pull/2666

closed 2021-11-27 by limonspb