betaflight/betaflight-configurator #2669: cli-tab can no longer be used

TheIsotopes, 2021-11-27:
after testing the last master on windows 10 I noticed that the cli-tab can no longer be used. The input line is continuously filled with characters and does not stop. If I reset the content of #2660, everything works again without any problems.

@haslinghuis pls can you check this.

haslinghuis, 2021-11-27:
@TheIsotopes Just tested master on Windows. That PR is only adding a 100ms timeout to allow MSP commands to complete when switching to CLI when coming from the PID Tuning tab. Also tested on Linux. How can I reproduce?

TheIsotopes, 2021-11-27:
@haslinghuis For example, I use a Matek F7 mini with default settings.
I'm going to cli execute a `diff` then `exit`, after reconnect going back to cli and it fills the input line with characters.

here is a screenshot, not quite two lines are filled with characters.
[image]

haslinghuis, 2021-11-27:
Interesting if you using remember last tab and being on the PID tuning tab it work's ok.
So we have to look at CLI here.

TheIsotopes, 2021-11-27:
Here another screenshot with console output ... hope this help you.

[image]

haslinghuis, 2021-11-27:
@TheIsotopes yeah intermittent saw this issue while in development, especially with latest firmware. Just tested with Matek F722 Mini on 4.2.11 and 4.3.
This PR only addresses going to CLI if on the PID Tuning tab. So this issue is not with this PR but another issue we have to address. 

Maybe we should stop sending new MSP commands if they get in retry state after some amount and bail out.

TheIsotopes, 2021-11-27:
so it is now more a problem with the firmware and less one with the configurator?

haslinghuis, 2021-11-27:
@TheIsotopes: The MSPCodes shown are from update_live_status() in serial_backend.js which are not supposed to run in CLI tab anyway. So I'm thinking having a zero timeout is the issue of this temporary fix.

haslinghuis, 2021-11-27:
@TheIsotopes Could you please test PR 2670 if this solves the issue for you.

TheIsotopes, 2021-11-27:
Tested now https://github.com/betaflight/betaflight-configurator/pull/2670/commits/2ebc1550d3894a82d2ddf95e6f7a8c75134faeb8 ... all looks good, the issue was solved.

thx @haslinghuis 👍

PR #2670 (2021-11-29): Make timeout run conditional instead of allowing 0
Fixes: #2669 

Builds: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=4660&view=artifacts&pathAsName=false&type=publishedArtifacts
closed 2021-11-29 by limonspb