betaflight/betaflight-configurator #2675: Configurator Installation consistently freezes at this specific step, across multiple machines

malinhiles, 2021-12-02:
## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem:
Betaflight Configurator setup wizard gets stuck consistently at this point in installation:
[image]

2. Include ways to reproduce the problem;
Perform a fresh install of the 10.7.1 configurator on a 64bit Windows 10 machine. I've had this same issue across 4 different machines this month

Kubiskoi, 2021-12-05:
Freezes for me also at exactly same step.

PeterE1968, 2021-12-05:
Yes same looking for me!

bluelegion, 2021-12-05:
Could be related to the "fortawesome" typeo in the screenshot above?

haslinghuis, 2021-12-05:
This problem is already fixed in 10.8.0.

KimIV, 2021-12-13:
На Виндовс 7 тоже "зависает". Я решил подождать и установка продолжилась. Общее время установки около часа.

truglodite, 2021-12-21:
I have seen the ole "this is fixed in 10.8.0"... problem being for the rest... releases top out at 10.7.1.

So "fixed in 10.8.0"... will be meaningless to those who don't have access to 10.8.

haslinghuis, 2021-12-21:
:santa:

ChadyG, 2021-12-30:
So the release for 1.0.7.1 should be disabled since it is not working?

YutakaMatsushige, 2022-01-06:
I confirm the installation of 10.7.1 on my computer is freezing in the same way at this point. I did not understand the reason and installed version 10.8

haslinghuis, 2022-01-06:
The 10.7.1 release was needed to work with 4.2.1x.

Asizon, 2022-01-07:
Fixed on 10.7.2

closed 2022-01-07 by Asizon