betaflight/betaflight-configurator #2680: Configurator is not able to Flash F3 and F1 FC anymore, also the older firmware files for this FS's are removed from Backend

MJ666, 2021-12-03:
Running Windows 11 and tested with BF Configurator 10.7.1 and 10.8 beta. I can flash F4 and F7 FC with no issues. If i try to flash a PikoBLX (F3), a SPRacinigF3Mini, a AlienflightF3 or a Naze32 (F1) it will always fail. An parallel install of BF Configurator 10.5.1 is able to flash all this FC's.

This is may be not the biggest problem to use an older BF Configurator but all F1 and F3 firmware files looks to be removed from the BF Configurator backend. They are still available on Git.

Many people will not replace this older FC's immediately and may have spare FC's to repair existing models. Even with the need to use an older BF configurator removing the firmware files from the backend make it very hard for users to repair there older gear.

Asizon, 2021-12-03:
Hello @MJ666 You can flash F1 and F3 disabling target selector, no board select in the dropdown, then select local file end flash it.

MJ666, 2021-12-03:
Many thanks for your feedback. this really makes sense to unselect the target board. It worked with 10.5.1 with an previous used target board still selected and flashing from a local file. It is also working now with 10.7.1 (actual latest release) when i unselect the target board. It is still not working for 10.8 beta. Tested this shortly with an AlienFlightF3 only.

Asizon, 2021-12-03:
Umm, is weird, i tested it with 10.8.0 some months ago and works, i will test again with my boards.

haslinghuis, 2021-12-03:
Could you show the console.log? F12

MJ666, 2021-12-04:
Here is the log from BF Configurator 10.8. Tried the same in 10.7.1 but F12 does not do anything?

[10.8.log](https://github.com/betaflight/betaflight-configurator/files/7653781/10.8.log)

haslinghuis, 2021-12-05:
```
firmware_flasher-24d49423.js:562 board changed to 0
firmware_flasher-24d49423.js:982 Loading file from: D:\Tools\Quadcopter\Settings - Firmware\Firmware\betaflight_4.0.6_ALIENFLIGHTF3.hex
firmware_flasher-24d49423.js:990 File loaded (719008)
hex_parser.js:104 HEX_PARSER - File parsed in: 59.286865234375 ms
serial.js:111 serial: connection opened with ID: 2 , Baud: 115200
MSPConnector.js:54 Connected
MSPConnector.js:70 Disconnected
serial.js:241 serial: closed connection with ID: 2, Sent: 6 bytes, Received: 9 bytes
serial.js:111 serial: connection opened with ID: 3 , Baud: 115200
stm32.js:133 Using legacy reboot method
stm32.js:135 Sending ascii "R" to reboot
serial.js:241 serial: closed connection with ID: 3, Sent: 1 bytes, Received: 0 bytes
port_handler.js:121 PortHandler - Removed: [{"path":"\\\\.\\COM10","displayName":"STM32 Virtual COM Port  "}]
```

It doesn't enter DFU mode here? Maybe try boot pins.

Asizon, 2021-12-05:
@haslinghuis F1 amd F3 has nlt DFU mode.

MJ666, 2021-12-06:
The F3 has DFU mode. The AFF3 is using the internal USB and is also entering DFU mode. At this point it gets stuck with 10.7.1. If i dort touch i can continue flash it with 10.5.1 as i also can flash with 10.5.1 completely. The SPRacigF3 is using an CP210x in front of the serial port and will use serial bootloader mode. This will stay in seral mode in BF Configurator.

haslinghuis, 2021-12-06:
@MJ666 Could you please try:

https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=4742&view=artifacts&pathAsName=false&type=publishedArtifacts

The flasher code does filter DFU mode for VCP and GD bootloaders.

MJ666, 2021-12-07:
Did the test but the behavior does not have changed. Here is the log:

[10.8new.log](https://github.com/betaflight/betaflight-configurator/files/7666025/10.8new.log)

MJ666, 2022-01-07:
Any news here? Anything useful in the log?

haslinghuis, 2022-02-08:
Reboot to DFU is not working:
```
serial.js:249 serial: closed connection with ID: 1, Sent: 2800 bytes, Received: 22121 bytes
FirmwareCache.js:234 Firmware cache journal loaded; number of entries: 0
FirmwareCache.js:234 Firmware cache journal loaded; number of entries: 0
firmware_flasher-24d49423.js:242 loaded some builds for later
firmware_flasher-24d49423.js:264 unified config cached data 4 mins old
firmware_flasher-24d49423.js:562 board changed to 0
firmware_flasher-24d49423.js:982 Loading file from: D:\Tools\Quadcopter\Settings - Firmware\Firmware\betaflight_4.0.6_ALIENFLIGHTF3.hex
firmware_flasher-24d49423.js:990 File loaded (719008)
hex_parser.js:104 HEX_PARSER - File parsed in: 50.02001953125 ms
serial.js:119 serial: connection opened with ID: 2 , Baud: 115200
MSPConnector.js:54 Connected
MSPConnector.js:70 Disconnected
serial.js:249 serial: closed connection with ID: 2, Sent: 6 bytes, Received: 9 bytes
serial.js:119 serial: connection opened with ID: 3 , Baud: 115200
stm32.js:133 Using legacy reboot method
stm32.js:135 Sending ascii "R" to reboot
serial.js:249 serial: closed connection with ID: 3, Sent: 1 bytes, Received: 0 bytes
port_handler.js:121 PortHandler - Removed: [{"path":"COM6","displayName":"STM32 Virtual COM Port ,"vendorId":1155,"productId":22336}]
```
@Asizon can you check with a F3?
@MJ666 or did we need to use `No reboot sequence` and put the board in DFU mode manually or using `bl` command in cli?

MJ666, 2022-02-09:
Mhhh, tested today with configurator 1.7.2 and 1.8.0 RC3 and a AlienflightF3 V2 FC. Flashing was working as it should both times at the first try. Not sure what has been changed since my last tests. I was even having the same results at as second computer that time. Now also on my laptop i flashed multiple FC with configurator 1.7.2 from different older versions to BF 4.0.6 (latest for AlienFlightF3). During my last tests the board was going into the DFU mode but the configurator was not going to flashing it for some reason.

haslinghuis, 2022-02-09:
https://github.com/betaflight/betaflight/issues/11334

closed 2022-03-20 by github-actions[bot]