betaflight/betaflight-configurator #2689: Missing Save button

CMartin249, 2021-12-16:
### Describe the bug
On latest nightly build/apk, configurator on Android appears to be missing the save button when in the motors tab. Tried to turn on/off bidirectional dshot via Android app cannot be saved.

### To Reproduce
Connect flight controller (in my case it was a HappyModel Mob5 ELRS) to Android phone (moto g(7) power - Android 10) running latest Betaflight configurator nightly APK. Open Motors tab and toggle bi-directional dshot. No way to save configuration.

### Expected behavior
A Save button would allow the change to be saved and be in effect when the FC is restarted. Don't have any other android devices to test and see if its specific to my phone.

### Setup
- Configurator version: 10.8.0-Debug-ae2ef20c

Asizon, 2021-12-26:
@CMartin249 can you send a screenshot of your issue??

CMartin249, 2021-12-27:
[image]

CMartin249, 2021-12-27:
Used latest nightly to get screenshot. A button tries to appear but cannot scroll down to see/select it

haslinghuis, 2021-12-27:
The button is enabled after a configuration change on this tab.
But on Samsung S7 I only see `Motor Direction` and `Save and Reboot` Button.

Asizon, 2022-01-07:
Thanks for your report @CMartin249
Can you try that build?
https://artprodeau1.artifacts.visualstudio.com/A3a9037bc-5464-4b2e-9e8d-ed1a5aa29705/ce1d2eeb-ce65-4b95-a02a-7fee6c006c71/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL0JldGFmbGlnaHQvcHJvamVjdElkL2NlMWQyZWViLWNlNjUtNGI5NS1hMDJhLTdmZWU2YzAwNmM3MS9idWlsZElkLzUzODEvYXJ0aWZhY3ROYW1lL2JldGFmbGlnaHQtY29uZmlndXJhdG9yLWFuZHJvaWQ1/content?format=zip

closed 2022-01-09 by haslinghuis

CMartin249, 2022-01-11:
Sorry for the delay, I was out of town. Tested the latest nightly and its working great! Thanks.