betaflight/betaflight-configurator #2691: Serial Port connection issue - windows10 (not english version)

bike797, 2021-12-17:
### Describe the bug

BetaFlight Configuator 10.8 

### To Reproduce

### Expected behavior

### Setup

- Configurator version:

```
PASTE THE OUTPUT OF 'diff' HERE
```

haslinghuis, 2021-12-19:
These are commits from 2020. Did you try latest nightly build? There will be some updates this week. Please be aware you need to update firmware if using latest nightly build.

bike797, 2021-12-19:
> These are commits from 2020. Did you try latest nightly build? There will be some updates this week. Please be aware you need to update firmware if using latest nightly build.

When a new version comes out this week, I will check the issue after installing it.
In addition, there is also a USB connection issue when installing the latest release version v20211212.4870 
  -I checked several versions
   . Configurator nightlies v20211212.4870 (connection issue)
   . Configurator nightlies v20201215.1347 (connection issue)
   . Configurator nightlies v20201213.1334 (connection issue)
   . Configurator nightlies v20201021.1106 (connection issue)
   . Configurator nightlies v2020200622.607 (connection OK)
   . Configurator nightlies v20200709.692 (connection OK)

[image]

[image]

daehani79, 2021-12-20:
감사합니다. 정확히 같은 문제가 있어 헤메던중이였습니다.
한글윈도우에서 확실히 문제가 있네요.

Thank you @bike797 .
I have exactly same issue in Windows10 korean.

bike797, 2021-12-20:
Linking to other repositories with similar issues with serial port connections. Could it be a clue to solving the problem?
- BlueJay Digital ESC firmware   https://github.com/mathiasvr/bluejay-configurator/issues/20

Asizon, 2021-12-20:
Sorry @bike797 bit there is not connection issue, all works as expected, revise your pc configuration.

Asizon, 2021-12-20:
You can read your linked bluejay issue comment here https://github.com/mathiasvr/bluejay-configurator/issues/20#issuecomment-897338612

Asizon, 2021-12-20:
@DusKing1 can you check if there is an issue with chinesse version please?

bike797, 2021-12-20:
Serial connection issue is not solved.
As mentioned in the link, if you change the language of Windows, serial connection problems occur.
- 
[image]

Asizon, 2021-12-20:
Looks like windows problem, not betaflight problem. Lets see what says dusking.

bike797, 2021-12-20:
Thank you for your kindness.
I'm not a software engineer, but I'd like to help with this.

Asizon, 2021-12-20:
Thanks to you for the help and report.

daehani79, 2021-12-23:
> Thanks to you for the help and report.

Would you check this issue of iNavFlight?
https://github.com/iNavFlight/inav-configurator/issues/1319#issuecomment-999657308

Downgrade nw.js version to 0.45.6 please.

Asizon, 2021-12-23:
We are in 0.59.0 version now, we cant downgrade to a very old version, this is not the solution.

daehani79, 2021-12-23:
> We are in 0.59.0 version now, we cant downgrade to a very old version, this is not the solution.

Yes, you're right.
It's not a permanaently solution, but it can be a temporary solution until official release.

Asizon, 2021-12-23:
Anyway, can you install lates development build and try it please?

bike797, 2021-12-23:
I tryed Configurator nightlies v20211223.5093, it still have a connection problem, too.

daehani79, 2021-12-23:
> Anyway, can you install lates development build and try it please?

@Asizon My bro helped changing nw.js version, it works greatly.
Sorry I'm not familiar with github. I linked.
https://drive.google.com/file/d/19cz2iuh5PvUGwIBiAq5JpvrXUYaFUBuh

Asizon, 2021-12-23:
Just go to gulpfile.js file and change version whete says nwjs, thought 0.42.6 is good election.

bike797, 2021-12-26:
Betaflight Configurator 10.8.0 RC1.0 still has compatibility issues.
It was also confirmed that the issue was resolved when building with NW 0.45.6.

If the NW version cannot be lowered to 0.45.6, is there any way to solve the compatibility issue while applying the current 0.59.0?

bike797, 2022-01-26:
I tried configuration rc2.0, but still have same issue.

haslinghuis, 2022-01-30:
So the issue started here?

https://github.com/betaflight/blackbox-log-viewer-nightlies/releases/tag/v20200622.608

kaduceus, 2022-02-02:
RC3.0 also had the same problem and it was not resolved.

mirrson, 2022-02-05:
I solved this issue by installing nwjs 0.60. With RC3.
I asked to a youtuber who did this method and he taught me this method.(Should I mention he's channel?)
1. download nwjs 0.60 (https://nwjs.io/downloads/)
2. copy all the components(not just the folder) to the betaflight configurator folder and overwrite.
3. done!

haslinghuis, 2022-02-06:
Test version here: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=5902&view=artifacts&pathAsName=false&type=publishedArtifacts

bike797, 2022-02-06:
> Test version here: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=5902&view=artifacts&pathAsName=false&type=publishedArtifacts

It works perfectly!!..