betaflight/betaflight-configurator #2701: feedforward jitter and thrust linearization GUI values mixed up in latest nightly build 6737c560

JAL513, 2021-12-20:
### Describe the bug

When setting feed forward jitter in GUI, it changes thrust linearization.  Thrust linearization shows a different value. 

Values are all confirmed correct in the CLI.

### To Reproduce

Update feed forward jitter and check thrust linear in the CLI

### Expected behavior

Update feed forward in the GUI should update feed forward jitter.  Thrust linear should be set with GUI as well.

### Setup

- Configurator version:
- 10.8.0 debug 6737c560

```
PASTE THE OUTPUT OF 'diff' HERE
```
# version
# Betaflight / STM32F411 (S411) 4.3.0 Nov  9 2021 / 20:25:08 (6ecd0d232) MSP API: 1.44
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id HAMO

# name: Diamond

# resources
resource MOTOR 1 B08
resource MOTOR 2 B07
resource MOTOR 3 B06
resource MOTOR 4 B10

# feature
feature -RX_PARALLEL_PWM
feature -RX_SPI
feature RX_SERIAL
feature TELEMETRY
feature LED_STRIP

# serial
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200

# beeper
beeper -RX_LOST
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# led
led 0 7,6::C:10
led 1 8,6::C:10

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 13 0 1300 1700 0 0
aux 3 35 1 900 1300 0 0
aux 4 36 2 1700 2100 0 0

# adjrange
adjrange 0 0 3 900 2100 29 3 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200

# vtx
vtx 0 5 0 0 1 1000 1325
vtx 1 5 0 0 2 1325 1650
vtx 2 5 0 0 3 1650 2000

# master
set dyn_notch_q = 250
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 350
set acc_calibration = 73,-39,-49,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = CRSF
set dshot_idle_value = 750
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_max_cell_voltage = 435
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 330
set vbat_scale = 111
set ibata_scale = 1175
set yaw_motors_reversed = ON
set small_angle = 180
set thrust_linear = 25
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_rssi_dbm_alarm = -95
set osd_vbat_pos = 10604
set osd_rssi_pos = 57
set osd_link_quality_pos = 10295
set osd_rssi_dbm_pos = 121
set osd_tim_1_pos = 33
set osd_tim_2_pos = 10647
set osd_flymode_pos = 10273
set osd_throttle_pos = 8417
set osd_vtx_channel_pos = 10625
set osd_crosshairs_pos = 10477
set osd_current_pos = 10638
set osd_mah_drawn_pos = 10633
set osd_craft_name_pos = 364
set osd_warnings_pos = 14634
set osd_disarmed_pos = 266
set osd_core_temp_pos = 8385
set osd_rate_profile_name_pos = 8353
set osd_pid_profile_name_pos = 8321
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set osd_stat_min_rssi_dbm = ON
set debug_mode = GYRO_SCALED
set cpu_overclock = 120MHZ
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5917
set vcd_video_system = NTSC
set gyro_1_sensor_align = CW270
set gyro_1_align_yaw = 2700
set rpm_filter_fade_range_hz = 100
set name = Whoopsie Diamond

profile 0

# profile 0
set vbat_sag_compensation = 100
set anti_gravity_gain = 0
set iterm_relax_cutoff = 3
set yaw_lowpass_hz = 0
set p_pitch = 94
set d_pitch = 101
set f_pitch = 251
set p_roll = 90
set d_roll = 88
set f_roll = 240
set p_yaw = 90
set f_yaw = 240
set d_min_roll = 66
set d_min_pitch = 75
set feedforward_averaging = 2_POINT
set dyn_idle_min_rpm = 100
set simplified_master_multiplier = 200
set simplified_i_gain = 50
set simplified_d_gain = 110

rateprofile 0

# rateprofile 0
set thr_mid = 0
set roll_rc_rate = 12
set pitch_rc_rate = 12
set yaw_rc_rate = 12
set roll_expo = 60
set pitch_expo = 60
set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 85
set throttle_limit_type = SCALE

# end the command batch
batch end

Asizon, 2021-12-21:
@JAL513 Thanks for the input, but you should test it with latest dev build for firmware. All works as expected.

closed 2021-12-21 by Asizon

JAL513, 2021-12-21:
@Asizon Thanks for following up. Just to make sure I understand, I shouldn't submit issues with nightly builds? Or was I just behind with nightly build I was running? 
I did try a later version and see the same issue. 
Just want to make sure I understand the process and help where possible.

haslinghuis, 2021-12-21:
@JAL513 Please when testing latest https://github.com/betaflight/betaflight-configurator-nightlies it is mandatory to flash with the latest firmware.

JAL513, 2021-12-21:
@haslinghuis Understood. I didn't see there was later versions when I submitted that issue. Apologies for that. I did test the latest last night and ran into the same issue plus some others. I'll try and submit those.

Asizon, 2021-12-21:
Yes @JAL513 for betatesting use always lates development builds on firmware and also Configurator side, dont forget to make full chip erase when flash!