betaflight/betaflight-configurator #2707: Windows 7 support please!

BrandonsBakedBeans, 2021-12-22:
### Is your feature request related to a problem? Please describe.
I assist a good number of users with tuning and I dont own any machines past win7 or chromeOS on ARM. I need new computers to continue supporting others

### Describe the solution you'd like
Win7 support

### Describe alternatives you've considered
Virtual Box, multi booting to linux. All of these are beyond my level of expertise and I give up quickly in frustration.

Please don't forsake us cheapo Betaflight users and experts!

Asizon, 2021-12-22:
Please contact with google, they remove windows7 support,we cant do anithing here sorry.

BrandonsBakedBeans, 2021-12-22:
> Please contact with Google, they removed Windows 7 support, we cant do anithing here sorry.

Google?! I'm confused now. Maybe for the ChromeOS support but I just want the old win7 support that was in the very first few builds of 10.8 and BBE 3.6 nightlies.

haslinghuis, 2021-12-22:
Please see: https://github.com/betaflight/betaflight-configurator#windows-users

BrandonsBakedBeans, 2021-12-22:
> Please see: https://github.com/betaflight/betaflight-configurator#windows-users

This is what I want to have fixed ;)

Asizon, 2021-12-22:
The only solution is that you create a repository on github and upload an unofficial version compatible with windows 7. We cannot support w7 as it would break other compatibility with current systems.

BrandonsBakedBeans, 2021-12-22:
> The only solution is that you create a repository on github and upload an unofficial version compatible with windows 7. We cannot support w7 as it would break other compatibility with current systems.

Is this something you can help me with? I am not really sure how to begin. All I want to do is be able to help all my friends that routinely come to me for for advice but now they're all flashing 4.3 and I'm clueless :facepalm:

Asizon, 2021-12-22:
Stopped by another no working things, we also support macOs and Linux systems.

Asizon, 2021-12-22:
You can start changing nwjs version in the code to 0.44.6 version and compile.

BrandonsBakedBeans, 2021-12-22:
> You can start changing nwjs version in the code to 0.44.6 version and compile.

Where might I find info on this? This is way beyond me but I want to see it happen

Asizon, 2021-12-22:
Install it and try if flight controller conects:
https://anonfiles.com/7fWdR136w0/betaflight-configurator-installer_10.8.0_win32_zip

BrandonsBakedBeans, 2021-12-22:
> Install it and try if flight controller conects: https://anonfiles.com/7fWdR136w0/betaflight-configurator-installer_10.8.0_win32_zip

This seems to be working. I can connect, flash and make adjustments in the true 4.3/10.8 fashion. Thank you greatly!

Asizon, 2021-12-22:
Really? Please send screenshot to confirm

BrandonsBakedBeans, 2021-12-22:
> Really? Please send screenshots to confirm

yes it seems to be working almost perfect. Just flashed and configured a toothpick with blackbox to test. Everything goes great except when the saving and rebooting it doesn't refresh the configurator: I have to press disconnect twice before it refreshes and I can press connect once more and continue. The attached picture shows "waiting for data" when I need to press disconnect twice to continue.
[image]
![blackbox 10 8](https://user-images.githubusercontent.com/68960197/147163189-b4de2ae3-a594-4cb0-9343-5f574c0654
[image]
b9.JPG)

Asizon, 2021-12-22:
@McGiverGim Seems that serial conection is working again and recognized in Windows 7 systems lol. I used nwjs 0.59.0 version here.

BrandonsBakedBeans, 2021-12-22:
> @McGiverGim Seems that serial conection is working again and recognized in Windows 7 systems lol. I used nwjs 0.59.0 version here.

It didn't want to connect until I tried manually and just entered \\.\COM10 for the port. That worked but not the first time... it also auto detected when I closed the Configurator and reopened while the drone was plugged in and it recognised the stm32 connection automatically.

Asizon, 2021-12-23:
@BrandonsBakedBeans everything is definitely not working the first time as it should, so we will not be able to include it in the menu.

swilkens, 2021-12-31:
As an alternative; you could run a virtual linux machine that does offer support for the newer releases?

It wouldn't cost anything, just some space on your hard drive.

DireCat, 2021-12-31:
I've installed the configurator on Win10 machine, then I just copied it's files back on my Windows 7 PC, launched it there and it still works! As mentioned above, there is a bug with COM port, you'll have to close and open the program again in order to reconnect.
It would be much better to allow the installation and show a warning message for Windows 7 users, instead of blocking it completely for no significant reason.

BrandonsBakedBeans, 2021-12-31:
> I've installed the configurator on Win10 machine, then I just copied it's files back on my Windows 7 PC, launched it there and it still works! As mentioned above, there is a bug with COM port, [...]

Thank you, this is the reply I've been waiting for... I 100% agree with you, there _should_ be an option for win7 but a warning during install. It's obvious that the Betaflight Configurator does work still on win7, but slightly bugged so support was dropped. Still think there should be an option (albeit non-supported option) for everyone!

KimIV, 2022-01-04:
Can't install on Windows 7 x64

[image]
[image]
[image]

Mode-1, 2022-01-28:
I can copy mrsslycat: "I've installed the configurator on Win10 machine, then I just copied it's files back on my Windows 7 PC, launched it there and it still works! As mentioned above, there is a bug with COM port, you'll have to close and open the program again in order to reconnect. It would be much better to allow the installation and show a warning message for Windows 7 users, instead of blocking it completely for no significant reason." - same here - works.
-> Win 7 / 64 is still up 'n running on millions of pc's. Why cut backward compatibility for no reason ? The 32bit core of windows is still present (even in win7/64) without changes which is good. 32bit code can be run on XP,Win7/8/10/11 - it can be done if willing and BF would reach many more pilots.

closed 2022-03-06 by github-actions[bot]

deezums, 2022-05-02:
I bet you undo commit https://github.com/betaflight/betaflight-configurator/commit/dd0b46bee8457e218aa0ebbaf1f2d2685bd5715c

and it'd work perfectly fine. Or, at least more than it does now, which is none. My evidence is people being able to grab an install from W10 and directly copy it over to W7 and have it work. 

So glad windows 7 support could be broken because other random people want to control which OS I use. Bunch of control freaks running around nowadays...

Can anyone post a dropbox or something to the newest install off a W10 machine? I don't want to try and make node.js work on 7.

Thanks, guys.

BrandonsBakedBeans, 2022-05-02:
> Can anyone post a dropbox or something to the newest install off a W10 machine? I don't want to try and make node.js work on 7.
I believe this issue got resolved by including a win64 portable install (just unzip and run)
[image]

deezums, 2022-05-02:
Thank you, that works!

@McGiverGim, thanks for pushing useless restrictive commits! Mind your own OS, and I'll mind mine, thanks. What's essentially a fancy serial terminal does NOT require any particular OS. Anyone saying otherwise, lololol