betaflight/betaflight-configurator #2710: Configurator doesnt work on Huawei P40 pro

insticprince, 2021-12-25:
### Describe the bug

Android configurator doesnt work on P40 pro. When you open the app it loads the wasp and when the wasp gone you have just the frame of the app, the "connect button" is grey and not accessable. the 3 dots next to it also not accessable and the 3 dashes on the other corner also not accessable.
### To Reproduce

Just run the app on a P40pro.
### Expected behavior

Not accessable buttons, no text at all, only the 3 dashes button connect button and 3 dots button that non of them fuction
### Setup

- Configurator version: not displayed but it 10.8.0-rc1

```
PASTE THE OUTPUT OF 'diff' HERE
```

Asizon, 2021-12-26:
@insticprince Can you try changing/updating your phone webview app please?

insticprince, 2021-12-26:
> @insticprince Can you try changing/updating your phone webview app please?

i cant find a way that allows me to change webview provider. i have installed webview from play store but i cant choose it in the dev settings.

Asizon, 2022-01-07:
@insticprince webview supported version is from 80, can you check your version i ndevelopper options?

closed 2022-02-15 by github-actions[bot]