betaflight/betaflight-configurator #2713: BF10.8.0 RC1 GUI not opening - multiple processes running

Canaill51, 2021-12-25:
### Describe the bug

When starting BF10.8.0 RC1 GUI, the application does not show up, but multiple processes are being created and running.
Indeed, I cannot kill all processes easily. I have to kill them one by one... and some are running again.

### To Reproduce

- On a fresh built machine running Windows 10 21H2, install BF10.8.0 RC1 GUI.
- Start the application as usual.
- Open task manager to check BF GUI processes running.
- Kill these processes.

I did try to start the application as a basic user and then, running as Administrator: Same issue.

Added a video explaining the trouble:
[https://youtu.be/G0nM2Xf5T-A](url)

### Expected behavior

BF10.8.0 RC1 GUI should be starting without any issue, and showing up.
One single process should run, as usual.

### Setup

- Configurator version:
BF10.8.0 RC1 GUI 

```
PASTE THE OUTPUT OF 'diff' HERE
```

haslinghuis, 2021-12-26:
What specs does your machine have. Specifically GPU?

Please take a look here and report back if this parameter solves your issue:
https://github.com/betaflight/betaflight-configurator/issues/2063

Asizon, 2021-12-26:
nah @haslinghuis not related to the graphical issues, i can reproduce it in my recent w7 tests, but it open without problem in my current w11 machine.

Canaill51, 2021-12-26:
> What specs does your machine have. Specifically GPU?
> 
> Please take a look here and report back if this parameter solves your issue: #2063

Thanks for your quick reply,
This machine is a Surface Pro 4 W11 21H2, CPU Intel Core I5-6300, GPU  Intel HD Graphics 520. Its graphic card driver is this one:
[image]

I just tried to test the workaround --disable-gpu, but still same issue.

Continuing to investigate, I updated its graphic card driver:
[image]

Same issue without --disable-gpu and with --disable-gpu.

I updated also chipset's update --> Same results.

What is strange: I had no problem with GUI 10.8.0 nightly build.
I have another machine running W11, no issue with the RC.

Are there any logs that BF GUI creates, that I can provide?

Canaill51, 2021-12-26:
I have found something interesting, I tried different nightly to reproduce this concern:
[image]

From Debug-4b8b2218, I have this problem. But all previous nightlies, all work fine.

Could I understand that this NW.js update makes the trouble?

ArimoIO, 2021-12-26:
sames issue here. Copied & overwrited the nw.js files from https://nwjs.io/ into the BF Folder. So far, it works now. Using Win10 on an Laptop with CPU n4100

Canaill51, 2021-12-26:
> sames issue here. Copied & overwrited the nw.js files from https://nwjs.io/ into the BF Folder. So far, it works now. Using Win10 on an Laptop with CPU n4100

Thanks for your tip, @ArimoIO !
Just did it and GUI is now showing up!

@Asizon @haslinghuis , please let us know what do you think about?

Asizon, 2021-12-26:
Umm, definitelly it can related to nwjs 0.59.0 update @haslinghuis ?

Asizon, 2021-12-26:
Yes, looking at release notes seems that there is a bug that doesnt open aplication in some operating systems, i will fix now reverting nwjs update

Asizon, 2021-12-26:
Thanks for the report guys, can youtry with this please?
https://artprodeau1.artifacts.visualstudio.com/A3a9037bc-5464-4b2e-9e8d-ed1a5aa29705/ce1d2eeb-ce65-4b95-a02a-7fee6c006c71/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL0JldGFmbGlnaHQvcHJvamVjdElkL2NlMWQyZWViLWNlNjUtNGI5NS1hMDJhLTdmZWU2YzAwNmM3MS9idWlsZElkLzUxNDAvYXJ0aWZhY3ROYW1lL2JldGFmbGlnaHQtY29uZmlndXJhdG9yLXdpbmRvd3M1/content?format=zip

TheIsotopes, 2021-12-26:
same problem here ... update to 0.59.1 and the issue is solved

Asizon, 2021-12-26:
can you try this please?
https://artprodeau1.artifacts.visualstudio.com/A3a9037bc-5464-4b2e-9e8d-ed1a5aa29705/ce1d2eeb-ce65-4b95-a02a-7fee6c006c71/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL0JldGFmbGlnaHQvcHJvamVjdElkL2NlMWQyZWViLWNlNjUtNGI5NS1hMDJhLTdmZWU2YzAwNmM3MS9idWlsZElkLzUxNDAvYXJ0aWZhY3ROYW1lL2JldGFmbGlnaHQtY29uZmlndXJhdG9yLWxpbnV40/content?format=zip

Asizon, 2021-12-26:
Yes, we revert the update to avoid more posible issues here that we dont know

Canaill51, 2021-12-26:
> Thanks for the report guys, can youtry with this please? [...]

Sure, I will test this yet, Of course :) Will let you know

Canaill51, 2021-12-26:
@Asizon , I confirm, your change fixed the problem. GUI can show up nice yet. I flashed a board without any issue, sliders seem working...

You all really rock :)

ArimoIO, 2021-12-26:
seems to work

haslinghuis, 2021-12-28:
@Canaill51 could you please also test with 59.1

Are there any users with M1 Mac here available for testing.

Canaill51, 2021-12-28:
> @Canaill51 could you please also test with 59.1
> 
> Are there any users with M1 Mac here available for testing.

@haslinghuis I previouly tried with 59.1 per @ArimoIO comment and I was able to get the GUI showing up too.
If you need more testing, let me know

iconx, 2021-12-31:
@Asizon brand new 11th gen i9, with other bells and whistles on win 10 pro. 
Exact same listed behavior, working with your provided update.
I had been using my phone for so long... finally a larger screen again.
Thank you homie. 😎 🌴

closed 2022-01-02 by limonspb