betaflight/betaflight-configurator #2717: Withe square around text in OSD

LoakAudio, 2021-12-26:
### Describe the bug
 When moving an item in the OSD tab, a large withe square appear around the item making it difficult to position

### To Reproduce
Click an try to move an item in the OSD Tab of Betaflight configurator
### Expected behavior

The withe part should be transparent

[image]

### Setup

- Configurator version:  Configurator: 10.8.0-debug-90f15faf / Debian 10 (Betaflight Configurator 10.8.0 Release Candidate 1)

```
PASTE THE OUTPUT OF 'diff' HERE
```
Maybe not needed ?

Asizon, 2021-12-26:
@LoakAudio It is intended to improve to facilitate positioning, I do not understand why it is more difficult for you now.

Asizon, 2021-12-26:
ouch, you are talking about the white square ? this is really weird!

LoakAudio, 2021-12-26:
Sorry "white" yes

Asizon, 2021-12-26:
can you try with this please?
https://artprodeau1.artifacts.visualstudio.com/A3a9037bc-5464-4b2e-9e8d-ed1a5aa29705/ce1d2eeb-ce65-4b95-a02a-7fee6c006c71/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL0JldGFmbGlnaHQvcHJvamVjdElkL2NlMWQyZWViLWNlNjUtNGI5NS1hMDJhLTdmZWU2YzAwNmM3MS9idWlsZElkLzUxNDAvYXJ0aWZhY3ROYW1lL2JldGFmbGlnaHQtY29uZmlndXJhdG9yLWxpbnV40/content?format=zip

LoakAudio, 2021-12-26:
@Asizon, Perfect with this betaflight-configurator_10.8.0-debug-4f415d75 version.
Thanks !
Can I use it and upgrade later with RC2 ?

Asizon, 2021-12-26:
thankyou for your report and test!

LoakAudio, 2021-12-26:
@Asizon an other question, the the "TX Link Power" doesn't appear in the OSD (the value is 0), my link is ELRS. Will I have to open a bug report for Betaflight or for ELRS ?

Asizon, 2021-12-26:
It should be report on firmware repo, please be sure there are not already opened issue.

LoakAudio, 2021-12-26:
Super efficient, thank you again, and still let go of your keyboard to have a good holiday!

Asizon, 2021-12-27:
Please keep it open, it will close automatically when fix is merged.

closed 2022-01-02 by limonspb