betaflight/betaflight-configurator #2723: To be able to connect the apk by the Speedybee adapter

LoakAudio, 2021-12-30:
### Is your feature request related to a problem? Please describe.
Depending on the generation of Android devices, multiple cables are needed to connect to a FC: USB C -> USB C, USB OTG + USB -> USB C, USB OTG + USB -> Micro USB. In addition USB OTG is not well implemented on every device (ex Fairphone or old Android devices)

### Describe the solution you'd like
Since it shows in Bluetooth list of all my devices, the Speedybee adapter seems to be an effective alternative to avoid cables and OTG, but I can't connect thru standard Bluetooth protocol. If the used protocol isn't proprietary, Can it be added to the rolling menu list of ports in the apk ?

### Describe alternatives you've considered
Continue to use a lot of cables

Asizon, 2021-12-30:
This is a feature request for future developments.

LoakAudio, 2021-12-30:
@Asizon ,
Yes, I understand that it is not a priority but in a more or less close future it would be interesting to be able to use Betaflight-Configurator by this device.

haslinghuis, 2024-06-23:
Duplicate of #3141

closed 2024-06-23 by haslinghuis