betaflight/betaflight-configurator #2725: Why it needs to remove old versions of Configurator?

rodrigosclosa, 2021-12-31:
### Is your feature request related to a problem? Please describe.
Why is it necessary to remove previous versions of the Configurator, even though it is an RC version? If this is a testing and unstable version, it shouldn't be mandatory to remove old versions (like 10.7.1) at install time and not have an option to let the user decide this.

### Describe the solution you'd like
Should not automatically remove. User should select whether to keep previous versions or not. For Beta or RC releases, we can continue using stable releases without any problems.

### Describe alternatives you've considered
Keep all versions by renaming directories and shortcuts with version numbers.

Asizon, 2021-12-31:
Dont worry, you can use 10.8.0 rc without problems for 4.2,no need to keep 10.7,remove it is mandatory for all computer software,also if it is not final version.

swilkens, 2021-12-31:
I like the suggestion.

[PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/releases), for example, lets users keep the latest stable installed alongside the latest -alpha, -beta and -RC. They do this by keeping a seperate installation location and data storage location for release versions and non-release versions.

An alternative for this would be to release the configuration as a stand-alone, not just an installer.

closed 2022-01-12 by haslinghuis