betaflight/betaflight-configurator #2731: Configurator hangs after flashing / save & reboot

Hellfire82, 2022-01-02:
### Describe the bug

All of these bugs seem to be related:

1) After successfully flashing the FC via DFU, the FC reboots but the configurator does not display a COM port (only options are 'Manual Selection' & 'Virtual Mode').
* There is a constant single red light on the FC which could be misjudged as the FC being stuck in DFU mode, however I don't think that this is the case (I believe this is because the custom defaults have not been applied).
* I have to restart the configurator in order for a COM port to be displayed.
* After connecting, the option to apply custom defaults is presented & the lights on the FC then display as normal.

2) In addition when selecting save & reboot, the FC reboots but the configurator hangs.
* I have to restart the configurator in order to return functionality.

3) In addition, when typing 'save' or 'bl' into the CLI, the FC reboots but the configurator hangs.
* I have to restart the configurator in order to return functionality (the latter with the FC in DFU mode).

[image]

### To Reproduce

1) Flash FC to any version (currently using 4.3 RC1)
* No COM port available so restart configurator.
* After restart, connect & apply custom defaults.

2) Select save & reboot.
* Restart configurator to return functionality.

3) Type save or bl into the CLI.
* Restart configurator to return functionality.

### Expected behavior

1) After successfully flashing the FC, the COM port should be displayed, which should then allow me to connect straight away & apply custom defaults.

2) When selecting save & reboot, the configurator should reboot & return to the previous tab that I was using.

3) When typing save or bl into the CLI, the configurator should reboot & return me to either the previous tab that I was using, or in respect to the latter, the home page.

### Setup

- Configurator version: 10.9.0-debug-90f15faf
- Windows 8.1
- Aikon F7 Mini v 1.1

* The ImpulseRC driver fixer has been ran on multiple occasions successfully.
* I installed configurator version 10.7.1 & all of the above issues were NOT present.

haslinghuis, 2022-01-02:
Please after flashing reconnect USB, this is mostly the case on F7 boards which would reset the FC. This PR should improve that: https://github.com/betaflight/betaflight/pull/11030.

Please press F12 in configurator to see the console. With that we can try to take a look.

There could be some problems with NW.js / Node upgrades. You could try this nightly which will be part of RC2: https://github.com/betaflight/betaflight-configurator-nightlies/releases/tag/v20220102.5278

Hellfire82, 2022-01-02:
I uninstalled the 10.8.0 RC1 configurator & installed the latest nightly build (Configurator: 10.8.0-debug-e19afcbf) & all of the aforementioned issues are still present.

Wasn't PR #11030 already merged into 10.8.0 RC1?

I'm not sure if this is what you need, but here's a screenshot of the console when attempting to enter the boot loader via the CLI:

[image]

It should be noted that when I load the configurator, the following error dialog box appears briefly before the configurator window appears:

[image]

Asizon, 2022-01-02:
This nwjs error is still normal since we have downgrade nwjs version, after flash issue is like @haslinghuis mentioned a known issue on F7 boards.

haslinghuis, 2022-01-02:
Just tried to tackle this on Windows 10, but all is working normal.
So a console message at the moment something is not working could give a clue.

Hellfire82, 2022-01-02:
I have just tried the same steps mentioned in my original post on an F4 FC and the same issues persist.

For some reason, every time I plug into a different FC, I have to run the ImpulseRC driver fixer before the COM port shows up in the configurator drop-down box. Also, if I have the configurator running when I plug in USB, no COM port appears until I restart the configurator.

It would seem I have an underlying problem with my laptop, so I am going to complete a fresh install of Windows in the hope that these issues are rectified.

haslinghuis, 2022-01-02:
Before you do also try a different USB port and data cable

Hellfire82, 2022-01-02:
Yes, I've tried 3 separate USB ports and 2 different data cables, which I know have worked previously and the results have been the same :(

Hellfire82, 2022-01-03:
I've performed a fresh install of Windows 8.1 & am currently using the latest nightly build of the configurator (10.8.0-debug-4d00fca6).
This has solved my issues with having to use the ImpulseRC driver fixer every time I connect a different FC, however the following issues still persist.

1) If the configurator window is loaded BEFORE the USB connection is plugged in, the COM port does not appear in the drop-down menu. The configurator has to be loaded AFTER the USB connection has been plugged in to make the COM port available. This issue occurs on both F4 & F7 FC's (both Aikon).

2) Whenever the FC is instructed to reboot from within the configurator i.e. save & reboot / bl in CLI etc. the configurator hangs. After restarting the configurator even though the COM port is still visible in the drop-down menu, you CANNOT connect. You have to shut down the configurator, physically unplug & reconnect the USB & then load the configurator to regain functionality. This issue occurs on both F4 & F7 FC's (both Aikon).

Let me know if there is anything I can provide to help you to understand / fix this issue.

haslinghuis, 2022-01-03:
Please try raising connection timeout on options tab. RC2 is coming soon. Please press F12 when something is not working so we might get a clue.

Hellfire82, 2022-01-03:
Raising the connection timeout did not resolve the issue, even to the highest value of 5000.

This is the output from the console when I save & reboot:

[image]

The data request time-out for ID's 68, 116, 150 & 110 recurs over & over.
I will wait for RC2 to see if that fixes this issue.

haslinghuis, 2022-01-03:
```
MSP_SET_REBOOT: 68
```
This MSP command should reboot the FC. This is obvious not executed here.

Hellfire82, 2022-01-13:
The following issues are still present, even after installing configurator 10.8.0 RC2.
When I tried configurator 10.7.2 everything worked as normal.

* Flashing - After flashing has completed, the configurator becomes unresponsive. I have to shutdown & restart the configurator. The COM port is then shown as normal & I can connect to apply the custom defaults.

* Save & Reboot - The configurator stays on the page where I pressed the save & reboot button. I have to press disconnect twice to get back to the welcome screen, then press connect. It should be noted that auto connect is turned on.

* Entering the boot loader via the CLI or the button in the setup tab - The configurator becomes unresponsive. I have to shutdown & restart the configurator. The FC is then shown to be in DFU mode as normal.

* Exiting DFU mode via the button on the firmware flasher page - The configurator becomes unresponsive. I have to shutdown & restart the configurator. The COM port is then shown as normal.

For some reason, the console is no longer displayed when I press F12, so I cannot provide any more screenshots of any errors that may be present.

haslinghuis, 2022-01-13:
Can you please test this version (it has NW.js v59 which was solving problems for Windows 7)

https://github.com/betaflight/betaflight-configurator-nightlies/releases/tag/v20211223.5093

Hellfire82, 2022-01-14:
I have installed the above version and all of the issues mentioned in my last comment are still present. The only change is that the console now works again, so I am able to provide screenshots for the different issues:

1) Flashing

[image]

2) Save and reboot

[image]

3) Entering the boot loader via the button in the setup tab

[image]

4) Entering the boot loader via the CLI

[image]

haslinghuis, 2022-01-14:
Do you have any peripherals connected on any of the UARTs (RX, GPS, VTX, ...) that are powered while you are trying to get into DFU mode? If so, disconnect them from the board, or disconnect the power to them, as they may interfere with the MCU binding to USB in DFU mode.
Also remove other USB peripherals which could hijack the port on reboot.

Hellfire82, 2022-01-14:
There is nothing else plugged into my laptop. Just the USB cable which is connected to my FC.

The only peripheral that I have connected on all of my FC's is my RX, which is powered by the 3v3 pad & therefore active when USB is plugged in. I do this so that I don't have to plug in a LiPo to update the RX's firmware. The VTX has no power when USB is plugged in. As I mentioned in a previous comment, this exact wiring setup works fine on configurator 10.7.2.

Configurator 10.8.0 is perfectly usable for me in it's current state. It just takes an extra few clicks / steps to achieve my goal. So if a fix is unable to be found without perhaps powering the RX from a 5V pad, then I will just continue like this.

haslinghuis, 2022-01-15:
Please test https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=5547&view=artifacts&pathAsName=false&type=publishedArtifacts

Hellfire82, 2022-01-15:
Unfortunately there is no change when selecting save and reboot. It still requires x2 clicks on disconnect, then x1 on connect.
The console errors are exactly the same as my screenshot regarding this issue.

haslinghuis, 2022-01-16:
Thanks. Guess indeed it has to do with additional hardware attached. It should look like:

[image]

Hellfire82, 2022-01-16:
Out of interest, I will try disconnecting the power wire for my RX to see if it makes any difference, if that is what you are referring to as hardware?
As previously mentioned, there are no other devices connected to my laptop by any means. Also, the same FC wiring connections work perfectly with 10.7.2, so a recent change in the configurator must be the issue.

Hellfire82, 2022-01-16:
As mentioned previously, the same FC wiring connections work perfectly with 10.7.2. The main difference between that version & 10.8.0 from my perspective is the move from x86 to 64bit. I don't have anything at all physically connected to my laptop other than the FC. However, I've just remembered that I am running a Windows virtual drive  & have an image permanently mounted. Could this have anything to do with these issues I'm having? I will remove this at the next opportunity to see if it makes any difference. If not, out of interest I will temporarily remove the 3v3 power wire for my RX to again see if it makes a difference.

haslinghuis, 2022-02-05:
@Hellfire82 Not able to reproduce the issue. Virtual environments could have issues encapsulating USB as I didn't check

Hellfire82, 2022-02-05:
Un-mounting the image didn't make any difference & all aforementioned issues are still present for me in RC3.
If you would like to remotely access my laptop to take a closer look for yourself, then let me know how we can accomplish this.

haslinghuis, 2022-02-06:
Please check https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=5902&view=artifacts&pathAsName=false&type=publishedArtifacts

Hellfire82, 2022-02-06:
After installation of the above version, nothing has improved from my perspective. As before, the USB connection has to be made before starting the configurator, or else the COM port doesn't appear in the port drop-down menu.
I have also tried a different cable to no avail.

1) Save & reboot - The FC reboots but requires 2 clicks of the disconnect button, then 1 click of the connect button to continue using the configurator.

[image]

2) Bootloader from CLI - The FC goes into DFU mode, but the configurator hangs & has to be restarted. When this is done, it shows the FC in DFU mode in the Port drop-down menu.

[image]

3) Save from CLI - The FC reboots but requires 2 clicks of the disconnect button, then 1 click of the connect button to continue using the configurator.

[image]

As previously mentioned, if you would like to access my laptop remotely to view these events more closely or to have a look at my settings etc. then let me know.

haslinghuis, 2022-02-06:
Debugging on Windows `clicks` is undefined (function serial_backend.js line 43). So this explains it's failing. Not sure why it still works here on my hardware and no more issues are filed.
```
    const thisElement = $(this);
    const clicks = thisElement.data('clicks');
```

haslinghuis, 2022-02-07:
Please test with: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=5914&view=artifacts&pathAsName=false&type=publishedArtifacts

Hellfire82, 2022-02-07:
All of the issues mentioned above are still present.

Save & reboot:
[image]

Bootlader from CLI:
[image]

haslinghuis, 2022-02-08:
Sorry forget to ask opening object by clicking on the arrow in the first image above.

McGiverGim, 2022-02-09:
> Debugging on Windows 10.0.19043 `clicks` is undefined (function serial_backend.js line 43). So this explains it's failing. Not sure why it still works here on my hardware and no more issues are filed.
> 
> ```
>             const thisElement = $(this);
>             const clicks = thisElement.data('clicks');
> 
>             console.log(clicks);
> 
>             const toggleStatus = function() {
>                 thisElement.data("clicks", !clicks);
>             };
> ```

I don't know if this is what you ask for, but `undefined` is similar to `false` for this case. `!undefinded == true` , so the first execution it is like `false`, next execution is `true`, etc.

Hellfire82, 2022-02-09:
@haslinghuis I wish it was hard for me to reproduce haha, it happens every single time.

Apologies for the delay. Here is the screenshot that you requested:

[image]

If you need me to expand and screenshot of any of the other lines of code within 'Object' then let me know.

haslinghuis, 2022-02-11:
@Hellfire82 thanks.

This draft should fix it: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=5940&view=artifacts&pathAsName=false&type=publishedArtifacts

Hellfire82, 2022-02-12:
@haslinghuis Unfortunately nothing has changed :(

After successfully flashing to 4.3 RC3:
[image]

Save & Reboot:
[image]

haslinghuis, 2022-02-13:
On F7 it is a known issue after flashing you have to reapply power (cold-boot).
See: https://github.com/betaflight/betaflight/pull/11030

But still the Aikon F7 FC here is not rebooting as it should. Do you have any other board to compare?

Hellfire82, 2022-02-13:
OK, I won't test when flashing again in that case.

I have two other FC's, an Aikon F4 and a Matek F411RX. They are both showing identical issues to that of my Aikon F7 as shown in the following screenshots. All of the above used to work perfectly on the 32-Bit version (10.7.2), so the issue must be related to something that has changed recently when moving to 64-Bit etc.

Aikon F4 save & reboot:
[image]

Matek F411RX save & reboot:
[image]

haslinghuis, 2022-02-14:
For testing: https://github.com/betaflight/betaflight-configurator/actions/runs/1838332428

Hellfire82, 2022-02-14:
Progress...
Upon clicking save & reboot, the FC reboots & I am taken back to the configurator welcome screen. It does not however take me back to the previous tab that I was using (auto connect is enabled).
When looking at the console prior to clicking save & reboot, it seems to be in a loop, outputting the same info over & over:

Save & reboot:
[image]
[image]
[image]

When typing save in the CLI, I still have to double click disconnect then single click connect.

Save via CLI:
[image]

haslinghuis, 2022-02-16:
Another way of testing in console while FC is connected you can simply type:
```
MSP.send_message(68);
```