betaflight/betaflight-configurator #2732: Roll axis mismatch between "Instruments" and "3d model"

matusfaro, 2022-01-02:
### Describe the bug
Under Setup tab, the "3d Model" does not match the "Instruments" in the Roll axis: When drone is physically rolled one way, the instruments roll one way and the model rolls the other way. See picture below.

I trusted the model for correct orientation and my drone flipped immediately. It seems the Instruments are correct. 

Mismatching roll axis:
[image]
[image]

Pitch axis is correct:
[image]
[image]

### To Reproduce
Using Lux H7 FC with two Gyros installed upside down in a drone.
Open Configurator Setup tab and view Instruments and 3d model.

### Expected behavior
I would expect the 3d model to match Instruments.

### Setup
- Configurator version: 10.7.1
- Target: MATEKH743
- Repro'd on BetaFlight versions: 4.2.9 and 4.3.0 Jan 1 2022 

```
# version
# Betaflight / MATEKH743 (M743) 4.3.0 Jan  1 2022 / 14:07:33 (1ceb80d11) MSP API: 1.44

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKH743
mcu_id 004c00423230510e33393233
signature 

# name: faro

# resources
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 B00
resource MOTOR 4 B01

# feature
feature MOTOR_STOP

# serial
serial 2 1 115200 57600 0 115200

# beacon
beacon RX_LOST

# map
map TAER1234

# aux
aux 0 0 1 1800 2100 0 0
aux 1 1 0 900 1200 0 0
aux 2 2 0 1300 1700 0 0

# master
set acc_calibration = -97,98,-3953,1
set max_check = 2000
set fpv_mix_degrees = 35
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set min_throttle = 1069
set motor_pwm_protocol = PWM
set align_board_roll = 180
set align_board_yaw = -90
set bat_capacity = 1300
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 16
set osd_vbat_pos = 14389
set osd_link_quality_pos = 14352
set osd_warnings_pos = 14569
set osd_battery_usage_pos = 14353
set osd_disarmed_pos = 14602
set debug_mode = GYRO_SCALED
set gyro_1_sensor_align = DEFAULT
set gyro_1_align_yaw = 2700
set gyro_2_sensor_align = DEFAULT
set name = matus

profile 0

profile 1

# profile 1
set p_pitch = 5
set i_pitch = 20
set d_pitch = 10
set p_roll = 5
set i_roll = 20
set d_roll = 10
set p_yaw = 20
set i_yaw = 20
set d_min_roll = 4
set d_min_pitch = 4

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 2

# save configuration
save
```

haslinghuis, 2022-01-02:
Are these default settings?
```
set align_board_roll = 180
set align_board_yaw = -90
set gyro_1_sensor_align = DEFAULT
set gyro_1_align_yaw = 2700
```

haslinghuis, 2022-01-02:
Please update configurator as firmware 4.3 does not work on configurator 10.7.x

Asizon, 2022-01-02:
This is user configuration issue, need to check correctly the board orientation @matusfaro

closed 2022-01-02 by matusfaro