betaflight/betaflight-configurator #2733: Motors stutter while idling (on arm)

dan1m1, 2022-01-03:
Hi please investigate following issue to see if can be reproduced on other quads
Thank you 

### Describe the bug

When arming the quad with the dynamic idle on set to 20 or below that the motors will stutter  (not running smooth) 

### To Reproduce

in the tuning tab go to dynamic idle and set it up to 20
in the motor tab leave the motor idle at default value
leave all PID tuning at Default
arm quad - notice that the motor are not running smooth, they stutter as when there is to much d term applied 
if you raise throttle the motors will start running smooth again
also if you increase dynamic idle value to 30 the motors will run smoother

### Expected behavior

Motors should run smooth at any value assign to dynamic idle field

### Setup

- Configurator version:
running kakute H7 flight controller with motors re-mapped to the second ESE header (motors 1,2,3,4 mapped to motor 5,6,7,8,)
using teeko 32 50 amp ESC and cyclefpv  25mm imperial motor 1870 kv on 6s 

```
# diff all

# version
# Betaflight / STM32H743 (SH74) 4.3.0 Dec 24 2021 / 18:48:24 (norevision) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEH7, version: e4b51680, date: 2021-10-04T00:52:01Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEH7
manufacturer_id HBRO
mcu_id 002a00443330511836353839
signature 

# name: Lumenier

# resources
resource MOTOR 1 A00
resource MOTOR 2 A02
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 B03
resource MOTOR 8 B10

# feature
feature -RX_PARALLEL_PWM
feature LED_STRIP

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# led
led 0 7,13::C:6
led 1 8,13::C:6
led 2 6,13::C:6
led 3 9,13::C:6
led 4 5,13::C:6
led 5 10,13::C:6
led 6 4,13::C:6
led 7 11,13::C:6
led 8 3,13::C:6
led 9 12,13::C:6
led 10 2,13::C:6
led 11 13,13::C:6
led 12 1,13::C:6
led 13 14,13::C:6
led 14 0,13::C:6
led 15 15,13::C:6

# aux
aux 0 0 1 1300 1700 0 0
aux 1 2 0 900 1225 0 0
aux 2 13 2 1300 2100 0 0
aux 3 35 2 1775 2100 0 0

# adjrange
adjrange 0 0 3 900 1325 12 3 0 0
adjrange 1 0 3 1375 1625 12 3 0 0
adjrange 2 0 3 1675 2100 12 3 0 0

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 700
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 475
set acc_calibration = 48,125,-201,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set rc_smoothing_setpoint_cutoff = 30
set rc_smoothing_feedforward_cutoff = 30
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_output_reordering = 1,3,0,2,4,5,6,7
set small_angle = 180
set deadband = 6
set yaw_deadband = 6
set simplified_gyro_filter_multiplier = 140
set osd_vbat_pos = 2544
set osd_current_pos = 2531
set osd_avg_cell_voltage_pos = 2538
set rpm_filter_q = 750
set name = Lumenier

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 82
set dterm_lpf1_dyn_max_hz = 165
set dterm_lpf1_static_hz = 0
set dterm_lpf2_static_hz = 165
set vbat_sag_compensation = 100
set iterm_relax_cutoff = 10
set yaw_lowpass_hz = 0
set throttle_boost = 9
set p_pitch = 59
set i_pitch = 106
set d_pitch = 47
set f_pitch = 126
set p_roll = 51
set i_roll = 92
set d_roll = 39
set f_roll = 110
set p_yaw = 51
set i_yaw = 92
set f_yaw = 110
set d_min_roll = 39
set d_min_pitch = 47
set auto_profile_cell_count = 6
set thrust_linear = 20
set feedforward_jitter_factor = 9
set dyn_idle_min_rpm = 23
set simplified_master_multiplier = 115
set simplified_d_gain = 115
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 80
set simplified_pitch_d_gain = 105
set simplified_pitch_pi_gain = 110
set simplified_dterm_filter_multiplier = 110

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 30
set thr_expo = 30
set roll_rc_rate = 23
set pitch_rc_rate = 23
set yaw_rc_rate = 26
set roll_expo = 69
set pitch_expo = 69
set yaw_expo = 69
set roll_srate = 83
set pitch_srate = 86
set yaw_srate = 83
set tpa_rate = 30

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

Above diff dose not have default tune any longer bit issue can be reproduces with any configuration on my quad

LoakAudio, 2022-01-03:
Hi @dan1m1 
My 2 cents, values between 30 and 40 are advised for dynamic idle in BF 4.3, if I trust the [tuning notes](https://github.com/betaflight/betaflight/wiki/4.3-Tuning-Notes#Improved-dynamic-idle)

dan1m1, 2022-01-04:
> 
Ok 
At value of 30 it is working fine there are no twitching detected in the motors.
If this is by design as per tuning notes this is not a bug however it feels that motors are idling at higher rpm that I was use to and feel that you should be able to reduce the dynamic idle below 30 without any issues (it worked in the previous nightly builds).
I will leave this open just a little longer for someone to acknowledge it and then I will close it as this might not be a bug at all.  

Thank you very much for the reply

phorensic, 2022-01-07:
I don't use dynamic idle, but I noticed in 4.3 my idle surges. Not really a stutter, more of an oscillation. This could be due to an overloaded FC with too much CPU usage, which I am diagnosing in another Issue. Did not do the idle surge in 4.2.

dan1m1, 2022-01-14:
No overloaded FC from what i can see in the tasks
it is not as bad as before, it looks that some how it resolved by itself  :) magic 
I will keep an eye on it

closed 2022-01-14 by dan1m1