betaflight/betaflight-configurator #2737: Cant change PID Loop Frequency

markyp1, 2022-01-04:
Flashed new firmware 4.3.0
Target IFRc/Iflight_745_aio_v2stm32f745)
Cant change pid loop frequency

### Setup

- Configurator version:10.8.0-debug-90f15af
- 

```
# version
# Betaflight / STM32F745 (S745) 4.3.0 Dec 24 2021 / 18:39:17 (norevision) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F745_AIO_V2, version: dba9876e, date: 2021-03-20T07:17:05Z

# start the command batch
batch start

board_name IFLIGHT_F745_AIO_V2
manufacturer_id IFRC

# master
set acc_calibration = -65,32,23,1

profile 0

rateprofile 0

# end the command batch
batch end
```

[image]

haslinghuis, 2022-01-04:
Guess you have a BMI270 which uses 3.6K. So PID loop will be restricted by that.

But according to https://cdn.sparkfun.com/assets/9/a/2/9/6/bst-bmi270-ds000.pdf it also has a 7.2K mode.

Asizon, 2022-01-04:
Yes Right, it seems BMI gyro, please close this isssue if it is.

closed 2022-01-07 by Asizon