betaflight/betaflight-configurator #2747: No osd in 4.3

davx35, 2022-01-09:
Hi there, dont know if there is a fix for this, but since updating to 4.3 rc2 I no longer can get osd to work.
Am using matek F722 fc with shark byte system and irc ghost rx. I get good image in goggles but no osd or stick controlls.
Any ideas as to a solution.

Cheers dave

haslinghuis, 2022-01-10:
Did you re-apply power to FC after configuring OSD? Did you upload Font? 
Please try latest dev build.
On my F411 it's working with 4.3RC2.

og54, 2022-01-10:
Check your osd_displayport_device and displayport_msp_serial settings. Will try rc2 myself tonight.

davx35, 2022-01-10:
I think the problem may be between 4.3 and hdzero/sharkbyte.

og54, 2022-01-10:
I've had no issues up to and including RC1, so unless there has been major alterations to canvas mode I don't think that's it

davx35, 2022-01-10:
I must admit that I was unaware that the OSD requires uploading in 4.3.

og54, 2022-01-10:
Just flashed RC2, works perfect on both my F4 and F7 HDzero quads. Check your osd_displayport_device and displayport_msp_serial settings.

davx35, 2022-01-10:
what am I looking for in there to be correct ?

og54, 2022-01-10:
set osd_displayport_device = MSP
set displayport_msp_serial = <UART# -1>. That means if you're on UART 4, it should be set displayport_msp_serial = 3 as an example

davx35, 2022-01-10:
Have identified the issue thanks to you. Had set MSP to uart 5 in the GUE, but in CLI was showing -1. One inserting this, set displayport_msp_serial = 4 everything is now fine. Many thanks again. dave Sent from Mail for Windows From: og54Sent: 10 January 2022 13:09To: betaflight/betaflight-configuratorCc: davx35; AuthorSubject: Re: [betaflight/betaflight-configurator] No osd in 4.3 (Issue #2747) set osd_displayport_device = MSPset displayport_msp_serial = <UART# -1>. That means if you're on UART 4, it should be set displayport_msp_serial = 3 as an example

SQuiDFPV, 2022-01-18:
HGLRC Zeus F722 - 1st quad updated to RC1 on release, NO OSD at all

today updated 2nd quad to RC2 - NO OSD at all. 
Cannot upload Font like the old days (greyed out)

CLI command im missing?

davx35, 2022-01-18:
Take a look at thread on github. Shows what you need to enter in cli. It worked for me !

closed 2022-01-18 by haslinghuis