betaflight/betaflight-configurator #2750: Adjustments tab; "then-apply" dropdown does not have "PID Profile Selection" (does have "PID-Audio Selection")

recyclerealestate, 2022-01-10:
### Describe the bug
On "adjustments" tab the "then-apply" drop down DOES NOT HAVE "PID-Profile selection" (but does have "PID-Audio Selection"?!?)

### To Reproduce
Go to adjustments tab and look at the "then-apply" dropdown list for "PID-Audio Selection"

### Expected behavior
Would expect a "PID Profile Selection" at the bottom of the list, next to "LED Profile Selection" and "OSD Profile Selection"
[I think I'd want the "Rate Profile Selection" also at the bottom of the list... I am not sure what order the options are in]

### Setup
- Configurator version: latest 10.8.0 RC2   10.8.0 (c9b7ed8f)

Just seems like a spell check changed "Profile" to "Audio"....
[image]
"

haslinghuis, 2022-01-15:
> Just seems like a auto-spell-correct changed "Profile" to "Audio"....

So using this `label `enables you to switch PID Profile?

recyclerealestate, 2022-01-15:
yes, in the same way as using the adjustments to switch the rate profile
with an aux channel.

haslinghuis, 2022-01-15:
Hmmm, who changed the label and why ;-)

haslinghuis, 2022-01-15:
Ah, it's not implemented: https://github.com/betaflight/betaflight/issues/2452

closed 2022-01-16 by haslinghuis