betaflight/betaflight-configurator #2757: No activate sliders button in PID tab (expert mode on). Can't turn filter toggles on, they revert to off when Save button engaged.

AlanzFPV123, 2022-01-15:
### Describe the bug
In PID tab, sliders are disabled with message to enable expert mode. However expert mode was already enabled. Also, in Filters tab toggles were both off. Turned them on and hit save but they reverted to off upon saving.

### To Reproduce
Don't know how you could, but I can upload the suspect hex file. See below.

### Expected behavior
Sliders may be deactivated if the Holybro custom 4,3 hex has pid changes (don't know if that's true), but regardless should be able to enable sliders and to toggle filters on/off. 

### Setup

- Configurator version: 10.8.0 RC1 and RC2

```
New build using Kakute H7. Flashed local hex from Holybro (Custom 4.3 dated 11/9/21). Confirmed with Holybro this is a tested hex. 
Issue first appeared using Configurator RC1. Updated to RC2 with identical behavior. Although guys on the UAV BF discord said RC2 fixed it for them. It did NOT fix the issue for me. Possibly a Holybro hex file issue, not compatible with RC1 or RC2? hex file predates these configurator builds. Just a layperson opinion there...

PASTE THE OUTPUT OF 'diff' HERE
```
# 

# Diff all

# version
# Betaflight / KAKUTEH7 (KTH7) 4.3.0 Nov  9 2021 / 11:26:18 (norevision) MSP API: 1.44

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEH7
mcu_id 001c003a3330511736353839
signature 

# name: CloudKicker7

# resources
resource MOTOR 1 B10
resource MOTOR 2 B03
resource MOTOR 3 B01
resource MOTOR 4 B00

# feature
feature GPS

# serial
serial 3 2 115200 115200 0 115200

# aux
aux 0 0 0 900 1100 0 0
aux 1 1 1 1900 2100 0 0
aux 2 2 1 1400 1600 0 0
aux 3 46 3 1900 2100 0 0
aux 4 13 2 1900 2100 0 0
aux 5 35 2 1875 2100 0 0

# master
set acc_calibration = 50,184,-35,1
set mag_hardware = NONE
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = WAAS
set simplified_gyro_filter = OFF
set simplified_gyro_filter_multiplier = 0
set osd_units = IMPERIAL
set osd_vbat_pos = 2476
set osd_rssi_pos = 2098
set osd_link_quality_pos = 50
set osd_rssi_dbm_pos = 83
set osd_tim_1_pos = 439
set osd_tim_2_pos = 407
set osd_flymode_pos = 2515
set osd_throttle_pos = 68
set osd_current_pos = 2118
set osd_craft_name_pos = 2434
set osd_gps_speed_pos = 2086
set osd_gps_lon_pos = 18882
set osd_gps_lat_pos = 18850
set osd_gps_sats_pos = 2102
set osd_home_dir_pos = 2159
set osd_home_dist_pos = 2125
set osd_altitude_pos = 18475
set osd_avg_cell_voltage_pos = 2508
set osd_disarmed_pos = 2347
set osd_flip_arrow_pos = 2273
set debug_mode = GYRO_SCALED
set name = CloudKicker7

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

haslinghuis, 2022-01-15:
Yes I can reproduce with the firmware which came with the board. However after flashing latest dev build it's working as expected.

AlanzFPV123, 2022-01-15:
Interesting. All I saw available were a couple 4.2.10 and 4.2.11 in the
configurator. Where can so get the latest dev build?

haslinghuis, 2022-01-15:
https://github.com/betaflight/betaflight-configurator/releases/tag/10.8.0-RC2

RC3 is around the corner

AlanzFPV123, 2022-01-15:
Thanks so much!

AlanzFPV123, 2022-01-15:
yet another issue - ugh. Kakute H7, I went back to 4.2.11 just to do a maiden as its a new build and not sure about that RC yet. I flashed using latest configurator RC2.  All seemed well until I noticed I had no OSD in the DJI goggles. Found that Uart 1 was not set to msp. when I try to enable msp, it will not save. This is true on laptop and also doesn't work using Speedybee app. Other changes I made saved with no issues
[image]
.

Asizon, 2022-01-15:
Closing this issue since it is an user configuration problem, please visit our slack for other support.

closed 2022-01-15 by Asizon

AlanzFPV123, 2022-01-15:
I don’t understand- changes not saving on ports tab in Configurator is a user configuration issue? I realize it’s separate from the original post/issue but not sure what to do at this point.