betaflight/betaflight-configurator #2760: 'Reboot to bootloader' boots to wrong bootloader.

hydra, 2022-01-17:
### Describe the bug
Betaflight supports bootloaders, bootloaders are used to do the flashing of firmware.  Two examples are a) the original OpenPilot bootloader, and b) the SPRacing EXST bootloader.

When the user clicks 'reboot to bootloader' they expect the firmware to reboot to the bootloader that allows for flashing new firmware via the tool they are currently using.  i.e.  use 'reboot to bootloader' in the configurator to flash via the configurator.  Users don't do anything else other than flash firmware or unplug the FC from the configurator when the FC is in it's bootloader mode.  Users in the SPRacing discord server will attest to this situation.

They do NOT expect 'reboot to bootloader' that takes them to a bootloader (the built-in ST ROM-based DFU bootloader) that can potentially require reflashing/recovery of the pre-installed bootloader.

The usual 'Flash Firmware' button reboots to the correct bootloader for flashing.

### To Reproduce
Click 'Reboot to bootloader', flash wrong firmware to FC that has a bootloader.

### Expected behavior
1) The firmware should reboot to the firmware used for flashing BY DEFAULT when asked to go to the 'bootloader' (both via CLI 'bl' command and MSP).
2) The configurator should reboot to the firmware used for flashing when 'reboot to bootloader' is pressed.
3) The firmware and configurator should NEVER boot to the ROM bootloader unless specifically requested, as in the case of an FC with a bootloader that is for developers and manufacturers only.  e.g.  they should use 'bl rom' in the CLI or jump the boot pads if they *really* need to, e.g. for connecting a debugger or flashing bootloader firmware, not BF firmware.

To summarize:
cli: `bl` should reboot to the 'default' bootloader.  i.e. flash-based bootloader, if one exists.
cli: `bl flash` should reboot to flash-based bootloader, if one exists.
cli: `bl rom` should reboot to ROM-based bootloader.
gui: 'reboot to bootloader' should reboot to the 'default' bootloader, i.e. flash-based bootloader, if one exists.
gui: 'flash firmware' should reboot to the 'default' bootloader, i.e. flash-based bootloader, if one exists.

gui: additionally, if there were 'reboot' and 'reboot to ROM bootloader' expert-user-only buttons in the GUI that would be helpful too.

### Setup
- Configurator version: nightly

```
N/A
```

### Additional information

Way back when the EXST bootloader existed the original behavior was to reboot to the DEFAULT bootloader, the target knows what it's default is.  In the case of normal targets it was ROM, in the case of EXST targets it was FLASH.

Then this PR was created 'Add booting into the flash boot loader as an option' which changed the default behavior to always be the ROM bootloader.  I disagreed strongly via slack at the time that this would cause support issues, additional documentation and a bad user experience.
https://github.com/betaflight/betaflight/pull/8429

Due to that change more code had to be added to the configurator to allow it to reboot to the correct bootloader when flashing - previously it 'just worked' by rebooting to the 'default' bootloader, i.e. the one that allows flashing.
https://github.com/betaflight/betaflight/pull/8680

NikoN-FPV, 2022-01-17:
yes i agree

haslinghuis, 2022-01-23:
In common_post.h in firmware on line 392 we have:
```
#if defined(USE_EXST) && !defined(RAMBASED)
#define USE_FLASH_BOOT_LOADER
#endif
```

Unless I'm missing something `USE_EXST` or `RAMBASED` are not defined anywhere?

hydra, 2022-01-23:
the targets define USE_EXST.  that setting is already enabled.  It's mostly the choice of the default bootloader in the firmware that is wrong.  The logic in the cli 'reboot' and msp 'reboot' commands is where the problem needs to be fixed.

haslinghuis, 2022-01-23:
Then we should transfer this issue to firmware first? Would you suggest reverting the PR's in question?