betaflight/betaflight-configurator #2762: H7 Board Extra UART "LPUART1" in Port Tab.

vincentpoont2, 2022-01-18:
### Describe the bug
Holybro Kakute H7 Board showing extra UART in ports tab. It is named "LPUART1"

### To Reproduce
Connect to Betaflight using Holybro Kakute H7 and go to port tab

### Expected behavior
This "LPUART1" should not appear

[image]

### Setup

- Configurator version: 10.8 RC2

```
dump

# version
# Betaflight / STM32H743 (SH74) 4.3.0 Jan  9 2022 / 12:14:26 (18ee33e7a) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEH7, version: e4b51680, date: 2021-10-04T00:52:01Z

# start the command batch
batch start

board_name KAKUTEH7
manufacturer_id HBRO

# name: -

# resources
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 B03
resource MOTOR 4 B10
resource MOTOR 5 A00
resource MOTOR 6 A02
resource MOTOR 7 C08
resource MOTOR 8 C09
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 D08
resource SERIAL_TX 4 D01
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
[... 1137 lines cut ...]
set roll_rc_rate = 7
set pitch_rc_rate = 7
set yaw_rc_rate = 7
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end

# 
```

haslinghuis, 2022-01-18:
LPUART1 is defined here for STM32_UNIFIED target: https://github.com/betaflight/betaflight/pull/10790

haslinghuis, 2022-01-20:
Asked around, but nobody seems to use the Low Power UART or knows any application.

vincentpoont2, 2022-01-20:
@haslinghuis I don't think it was turned on in the KakuteH7 Config file, so I guess it was turned on by default? Most H7 board do not have this port, may I ask how to remove it for KakuteH7?

closed 2022-01-25 by haslinghuis