betaflight/betaflight-configurator #2763: Selected preset is not loading correctly

elvisteck, 2022-01-19:
### Describe the bug
After selecting a preset, the window that opens to select the options does not load correctly

[image]

### To Reproduce
Enter the presets tab and select the supaflyFPV Freestyle 5" tuning preset. 
2 days ago it worked fine for me

### Setup
SO: Windows
Configurador: 10.8.0 (c9b7ed8f)
```
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jan  9 2022 / 11:51:40 (ee8560af4) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_SUCCEX_E_F7, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

# start the command batch
batch start

board_name IFLIGHT_SUCCEX_E_F7
manufacturer_id IFRC

# name: Nazgul

# serial
serial 0 1 115200 57600 0 115200

# beacon
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 1325 1675 0 0
aux 2 2 1 1750 2100 0 0
aux 3 13 4 1750 2100 0 0
aux 4 35 2 1800 2100 0 0

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 2
set dyn_notch_q = 450
set dyn_notch_min_hz = 100
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = -11,16,-53,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 12
set serialrx_provider = CRSF
set dshot_bidir = ON
set gyro_1_align_yaw = 1800
set name = Nazgul

profile 0

rateprofile 0

# end the command batch
batch end
```

Asizon, 2022-01-19:
@elvisteck yes we have made some changes in presets, can yo try latest development build?

elvisteck, 2022-01-19:
Hi @Asizon I tested the version v20220118.5652 (https://github.com/betaflight/betaflight-configurator-nightlies/releases/tag/v20220118.5652) and it's ok the window of official presets
[image]

Note: I changed to the Backup source of presets and appears the follow message. I think this its ok because is the source of the community and might have some errors
[image]

limonspb, 2022-01-19:
> Hi @Asizon I tested the version v20220118.5652 (https://github.com/betaflight/betaflight-configurator-nightlies/releases/tag/v20220118.5652) and it's ok the window of official presets [image]
>
> Note: [...]

The backup branch has not been updated to the latest format yet. So undefined.undefined version is normal there in that window.

There official Betaflight source must work ok without this window on the nightly configurator.

closed 2022-01-19 by elvisteck