betaflight/betaflight-configurator #2773: Progress bar becomes gray after firmware update

bakwc, 2022-01-26:
### Describe the bug

After upgrading firmware - progress bar expected to stay green. And now it becomes gray automatically.

### To Reproduce

* Update Firmware
* Progress bar become green
* Progress bar become gray after a second

### Expected behavior

* Update Firmware
* Progress bar become green
* Progress bar stay green

### Configurator version

10.8.0 (c9b7ed8f)

Asizon, 2022-01-26:
Can you send a screenshot please?

bakwc, 2022-01-26:
I can do better, send a video: https://drive.google.com/file/d/1YeLUX4hDoCvuFUnFnImFvV57lMwogkNf/view?usp=sharing
00:46 progress bar is green
00:49 progress bar is gray
I didn't press anything.
I used progress bar color to know if I already flash my FC or not. And now I sometimes flash it multiple times.

haslinghuis, 2022-01-26:
Somehow refresh is triggered on Windows reloading the tab. This does not happen on Linux.

bakwc, 2022-01-26:
On MacOS.

haslinghuis, 2022-01-26:
Please check https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=5775&view=artifacts&pathAsName=false&type=publishedArtifacts

bakwc, 2022-01-27:
Thank you! Fixed.

closed 2022-01-27 by bakwc