betaflight/betaflight-configurator #2774: Betaflight  is not pulling up presets

addy911, 2022-01-26:
### Describe the bug

So when you click on a preset it is not loading them completely it's not letting me actually click " pick "and then agree and flash .. instead like on superflies where there's option a drop-down bar to pick which option you want it just stays white and then it doesn't flash it.. and I just got a new flight controller in yesterday I'm not sure if this is the flight controller or betaflight but whenever I save my modes I unplug it will arm the first time but. After I power the quad off my modes reset

### To Reproduce

Described above

### Expected behavior

Normally the presets load up and flash no problem and my modes are remembered.

### Configurator version

10.8.0 release 2

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

### Add any other context about the problem that you think might be relevant here

I'm using the infinity stack 30x30 but I also have a friend of mine that's using a different stack  and it's happening to him to

haslinghuis, 2022-01-26:
Sorry, this is expected: we currently need nightly builds as things needed minor modification. RC3 is around the corner.

addy911, 2022-01-26:
Awesome thank you!!

closed 2022-01-26 by haslinghuis