betaflight/betaflight-configurator #2784: DYNAMIC_FILTER switch disappears on the Configuration tab after upgrading to BF 4.3

StevieJ123, 2022-01-29:
### Describe the bug

Hi,
I've tried this on my iFlight Mini F7 TwinG and also Holybro Kopis 2.5" Cinewhoop (KAKUTEF4V2 board), and in both cases I had the option to switch on the DYNAMIC_FILTER option in BF 4.2.x (v10.8 of BF Configurator), but on flashing them to BF 4.3 RC2, the switch has simply disappeared. I have powered on both quads, and both have BLHeli 32 ESCs. I have enabled Bidirectional DShot (600 on one, 300 on the other) but there is no option to enable the dynamic filter. I want to be able to use this filter on my tune. I think this must be a bug because the option was available on both quads before flashing to BF 4.3 RC2. Please help.

### To Reproduce

1) Update firmware to BF 4.3 RC2 using safe configuration items from previous diff -all.
2) Calibrate gyro and check/set appropriate settings if any important ones haven't been ported over.
3) Enable bidirectional Dshot under the Motors tab
4) Set the number of motor poles to 12 (counted) under the Motors tab

### Expected behavior

DYNAMIC_FILTER should be visible and selectable on the Configuration tab within BF Configurator for BF 4.3 RC2.

### Configurator version

10.8

### Flight controller configuration

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff -all

# version
# Betaflight / STM32F405 (S405) 4.3.0 Jan  9 2022 / 11:40:24 (ee8560af4) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEF4V2, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name KAKUTEF4V2
manufacturer_id HBRO

# name: Kopis Freestyle

# feature
feature -AIRMODE
feature MOTOR_STOP
feature LED_STRIP

# serial
serial 5 1 115200 57600 0 115200

# beeper
beeper -ARMED

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TREA1234

# led
led 0 4,5::C:12
led 1 5,5::C:12
led 2 6,5::C:12
led 3 7,5::C:12
led 4 8,5::C:12
led 5 9,5::C:12

# aux
aux 0 0 1 1325 2100 0 0
aux 1 1 2 900 1325 0 0
aux 2 13 0 1650 2100 0 0
aux 3 28 3 1475 2100 0 0
aux 4 35 2 1625 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 83,96,-121,1
set serialrx_provider = CRSF
set dshot_idle_value = 400
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_max_cell_voltage = 435
set vbat_scale = 111
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set osd_vbat_pos = 450
set osd_rssi_pos = 14806
set osd_link_quality_pos = 14798
set osd_tim_2_pos = 14807
set osd_flymode_pos = 14733
set osd_current_pos = 459
set osd_warnings_pos = 329
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5740
set gyro_1_align_yaw = 2700
set name = Kopis Freestyle

profile 0

rateprofile 0

# end the command batch
batch end

haslinghuis, 2022-01-29:
@StevieJ123 the switch has moved: https://github.com/betaflight/betaflight-configurator/pull/2653

StevieJ123, 2022-01-29:
Many thanks for the quick response, that's awesome :-)

closed 2022-01-29 by StevieJ123