betaflight/betaflight-configurator #2789: Can't scroll down the Keywords dropdown within Presets

cynfewl, 2022-02-01:
### Describe the bug

If you dropdown the Keywords selection list under Presets, you can't see anything other than the initially displayed list.

Trying to scroll down flashes the screen, but the list does not move.

[image]

Other dropdowns on this page seem to work fine.

### To Reproduce

1. Connect to a flight controller.
2. Navigate to Presets section.
3. Drop down Keywords selection list.
4. Attempt to scroll up or down using mouse wheel or sliders.

### Expected behavior

Keywords list should scroll up and down and allow you to select filter options as it did in previous version.

### Configurator version

Configurator: 10.8.0 (340087a4)

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

Asizon, 2022-02-02:
@cynfewl working fine for me...seems an issue in your computer, which SO are you using?

Asizon, 2022-02-02:
OK, i can reproduce now,good catch @cynfewl !

Asizon, 2022-02-02:
itis just happens with keyword an using official presets @limonspb , so the is some broken on firmware presets repo?

limonspb, 2022-02-02:
please try this Configurator and report if it works good now:
https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=5882&view=artifacts&pathAsName=false&type=publishedArtifacts

cynfewl, 2022-02-02:
That's an impressive turnaround!

Tested with portable and installed version and it is indeed fixed (or should I say hack-fixed :) ).

Thank you very much! The new presets rule.

haslinghuis, 2022-02-09:
Fixed in: https://github.com/betaflight/betaflight-configurator/pull/2790

closed 2022-02-09 by haslinghuis