betaflight/betaflight-configurator #2793: NW.js error dialogue on startup

jehe79, 2022-02-02:
### Describe the bug

[image]
Error dialogue presented when starting RC3 in Ubuntu - dissapearing as soon as BF configurator has started.

Text (in swedish) says "Your profile can't be used since it was created with a newer version of NW.js. Some functions might not be accessible. Use another profile directory or use a newer version of NW.js.".

I've had no issue using the configurator this far, only flashing and restoring one quad.

### To Reproduce

Close and restart Betaflight configurator.

### Expected behavior

No error message on startup.

### Configurator version

10.8.0 (340087a4)

### Flight controller configuration

N/A

### Add any other context about the problem that you think might be relevant here

Probably only a Linux/Ubuntu issue.
Installed betaflight-configurator_10.8.0_amd64.deb using Ubuntu 20.04.

Previously had RC1 so this is most likely happening because of:
https://github.com/betaflight/betaflight-configurator/pull/2716
.. introduced in RC2.

KarateBrot, 2022-02-02:
I fixed it like this:
- Uninstall Configurator
- delete all files that are left in the Configurator folder
- Install Configurator RC3

jehe79, 2022-02-03:
Yes - problem can be solved by removing / renaming:
/home/$USER/.config/betaflight-configurator

Anyhow - I'd probably like to know what might be stored in my profile, so I know what I'd be wiping.

McGiverGim, 2022-02-03:
In the profile, it stores the "configuration" of the Configurator. For example if you have selected "expert" mode or not, language selected, etc.

Asizon, 2022-02-14:
Fixes in https://github.com/betaflight/betaflight-configurator/pull/2797

closed 2022-02-14 by Asizon