betaflight/betaflight-configurator #2799: RC3 : "Preset source version mismatch" pop-up error when trying to load my repo

AndroidGX, 2022-02-08:
### Describe the bug

Hello,
I'm currently with latest available configurator (RC3) for BF 4.3, and since I updated from RC2 to RC3, I now have the following error when it tries to load my preset repo : 

[image]

I tried to reinstall RC3 just today but same issue.
Any idea how to fix that ?

Many thanks in advance for the help.

### To Reproduce

Try to load my own github repo while beeing in RC3.
Link of my repo : https://github.com/AndroidGX/firmware-presets/tree/Feisar-4.3

### Expected behavior

No warning pop-up, like in RC2.

### Configurator version

340087a4

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

McGiverGim, 2022-02-08:
I think you need to add the version to your repo, like in the Betaflight presets repo:

https://github.com/betaflight/firmware-presets/blob/005203122a15e05d80478bd35a0ea9214f152996/index.json#L1-L3

haslinghuis, 2022-02-08:
This is for RC2 but should apply to RC3:

[image]

AndroidGX, 2022-02-08:
> I think you need to add the version to your repo, like in the Betaflight presets repo:
> 
> https://github.com/betaflight/firmware-presets/blob/005203122a15e05d80478bd35a0ea9214f152996/index.json#L1-L3
> 
> Probably you are using an old version of the repo. The Betaflight repo has evolved since you forked it and added this values.

Many thanks, I deleted my fork and recreated a new one and it works perfectly this time 👍

closed 2022-02-08 by AndroidGX

KarateBrot, 2022-02-08:
@AndroidGX You don't need to delete it. You can always pull the changes into your fork to update