betaflight/betaflight-configurator #2806: Preset "Loadouts"

ItsBlunty, 2022-02-12:
### Is your feature request related to a problem? Please describe

The preset system is amazing and holds a lot of options that are quickly accessible to users, but in the current implementation there are still guides and generally guidance needed to help users decide what presets to use and what needs to be applied.

### Describe the solution you'd like

I feel that if additional UI elements were added at the top of the Presets page to allow selecting multiple presets at once across a few predefined categories it would encourage and show the user what they should be aiming for when picking presets without direct tutorialization.

For instance, at the top you would have dropdowns such as "TUNE" "RC_LINK" and "RC_SMOOTHING". These seem like the big 3 that each user might want to apply.

Each of these would be filled at the start with each categories' Betaflight 4.3 Defaults.

Then the user could select their options for each (picking a preset "loadout"), so they might pick "SupaflyFPV Freestyle 5 Inch" "ExpressLRS 150hz" and "Cinematic" and then they could apply that and it would apply all the items at once.

### Describe alternatives you've considered

This may be outside of the goals of the preset system so hopefully something to this effect will line up with the vision of presets. Also maybe someone else could suggest more or a different set of Categories to list for loadouts, this is just based on my understanding of what most users would want to apply.

Also, if this is implemented I could imagine a future version where we have only this new "loadout" system showing unless the user selects the Expert mode tab.

TehllamaFPV, 2022-02-12:
First, I really like this idea. 
Being able to have some memory retention with this behavior would also be pretty excellent in my mind (e.g. remembering what presets were used last for each of these).  If it can retain the search capabilities in each, I think that would be a pretty solid way of approaching it.

I would suggest that  'RATES' 'FILTERS' 'TUNE' 'RC_LINK' 'RC_SMOOTHING' are the proper requisite choice, unless we shift towards 'TUNE' being a more inclusive category that implicitly carried as filter integration option as part of the drop-down for every tune setup (even if it's including the basic tunes).

We would need to have a 'null' option at the bottom, basically a 'leave what I have alone', but it would need to be at the bottom and prompt a disclaimer of 'No Preset of this type selected', but that would be a good way to allow users to apply a subset of these.

I think a simplified implementation might be the step towards a 'wizard' mode, with the addition of verifying board orientation and motor direction/idle speeds as well.

As manufacturers realize that BNF 'back to stock' presets make sense, for a lot of users this would be the only requisite configurator interaction - verifying board direction, motors direction, and applying this minimal set would be plenty, and we would have a true 'non-Expert' mode (and the current non-expert mode would become intermediate mode).
I don't know if it would live as a replacement of the Presets tab, or join the 'Setup' tab at the very front and allow for launching and selecting presets indirectly from there (and allow users to only use that one tab in the configurator unless they're building).  From a UX standpoint, I can see this being very valuable to hardware manufacturers.

limonspb, 2022-02-12:
@ItsBlunty not sure if I understand the idea correctly. But some of it sounds reasonable, so i want to know more.

we already have "TUNE" "RC_LINK" and "RC_SMOOTHING" categories a user can select and search. Are you talking about some type of wizard?

Could you please provide a couple of dummy UI-pictures illustrating what you are talking about?
They should not be pretty, just in mspaint :) So that it was easier for me to imagine what you are talking about.
Example
[image]

ItsBlunty, 2022-02-13:
Here's my terrible paint rendition of my idea. Let me know if it something is confusing about this.

Just to reiterate, the goal here in my eyes is to present the user with a clear set of choices which directs them through the way the UI is setup to clearly pick one of each option and apply them, thus doing a basic preset "loadout".

[image]

limonspb, 2022-02-13:
> @limonspb Here's my terrible paint rendition of my idea. Let me know if it something is confusing about this.
> 
> Just to reiterate, the goal here in my eyes is to present the user with a clear set of choices which directs them through the way the UI is setup to clearly pick one of each option and apply them, thus doing a basic preset "loadout".
> 
> [image]

looks reasonable. Any idea on how a user would get to this screen? Like a separate tab or a button somewhere? How would you name it?

TehllamaFPV, 2022-02-15:
I had originally thought if there was a way to shove this under the 'Setup' tab, with links to the motor numbering/direction wizard also there.
[image]