betaflight/betaflight-configurator #2823: Motor poles setting not saving.

kaos-1, 2022-02-19:
### Describe the bug

Unable to save motor poles 

### To Reproduce

manually or using arrows set the motor poles, then save

### Expected behavior

should save and display on restart

### Configurator version

10.8.0 rc3 

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

haslinghuis, 2022-02-19:
Tested with V1 it works (don't have a V2 sample):
```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Feb  1 2022 / 20:14:54 (3267f0417) MSP API: 1.44
# config: manufacturer_id: SPBE, board_name: SPEEDYBEEF7MINI, version: 8e2d8dd4, date: 2021-11-30T04:01:12Z

# start the command batch
batch start

board_name SPEEDYBEEF7MINI
manufacturer_id SPBE

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
```

moonberry-pl, 2022-02-20:
The problem occurs when we use an older version of Betaflight, eg 4.2.11

Asizon, 2022-02-20:
No issue here, tested 4.2 and works good. Witch operating system???

Asizon, 2022-02-20:
Yes, thanks guys, fixed in https://github.com/betaflight/betaflight-configurator/pull/2817
You can try last nightly.

closed 2022-02-20 by Asizon