betaflight/betaflight-configurator #2849: vtx_power is not saved

palgy, 2022-03-04:
### Describe the bug

vtx_power is not saved , when setting from configurator or from cli 

### To Reproduce

update vtx power from configurator and save or   from cli 
set vtx_power = 5 
save 
disconnect the flight controller  from USB wait 10 sec 
check the vtx_power ... value was not saved 

### Expected behavior

vtx_power should be saved 

### Configurator version

10.8.0 RC3

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

### Add any other context about the problem that you think might be relevant here

defaults nosave

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Feb  1 2022 / 20:14:54 (3267f0417) MSP API: 1.44
# config: manufacturer_id: SPBE, board_name: SPEEDYBEEF7V2, version: 55694498, date: 2021-05-15T13:56:51Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name SPEEDYBEEF7V2
manufacturer_id SPBE
mcu_id 0030002f3756501120383558
signature 

# name: AOS 5

# feature
feature -ESC_SENSOR
feature GPS
feature TELEMETRY

# serial
serial 1 8192 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1775 2100 0 0
aux 1 1 2 1325 1650 0 0
aux 2 27 1 1775 2100 0 0
aux 3 13 4 1750 2100 0 0
aux 4 35 3 1750 2100 0 0

# vtxtable
vtxtable bands 7
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE  L CUSTOM  5333 5373 5413 5453 5493 5533 5573 5613
vtxtable band 7 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 5
vtxtable powervalues 10 2 25 100 400
vtxtable powerlabels 10 RCE 25 100 400

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -94,18,-73,1
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set motor_poles = 12
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1300
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set report_cell_voltage = ON
set osd_cap_alarm = 1300
set osd_vbat_pos = 2305
set osd_rssi_pos = 129
set osd_link_quality_pos = 2209
set osd_rssi_dbm_pos = 2177
set osd_tim_1_pos = 2401
set osd_tim_2_pos = 2369
set osd_flymode_pos = 2381
set osd_throttle_pos = 2361
set osd_vtx_channel_pos = 2497
set osd_current_pos = 2391
set osd_mah_drawn_pos = 2273
set osd_craft_name_pos = 2081
set osd_gps_speed_pos = 2264
set osd_gps_lon_pos = 2129
set osd_gps_lat_pos = 2097
set osd_gps_sats_pos = 2167
set osd_home_dir_pos = 2190
set osd_home_dist_pos = 2200
set osd_flight_dist_pos = 2232
set osd_compass_bar_pos = 2058
set osd_altitude_pos = 2327
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2337
set osd_disarmed_pos = 2411
set osd_esc_tmp_pos = 2465
set osd_flip_arrow_pos = 2113
set osd_core_temp_pos = 2296
set osd_log_status_pos = 2145
set osd_efficiency_pos = 231
set debug_mode = GYRO_SCALED
set vtx_band = 3
set vtx_channel = 4
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5645
set name = AOS 5

profile 0

# profile 0
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 45
set feedforward_jitter_factor = 6

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration

etracer65, 2022-03-09:
The `vtxtable` you have set up is wrong. The power levels section is wrong and that's why the power level can't be set. Contact the vtx manufacturer to determine which settings to use.

closed 2022-03-09 by etracer65

palgy, 2022-03-26:
Hi again., Iload the preset for openVTx Jye Smith.
with the preset's without any changes the power is still not saved

RadikalBanal, 2023-07-26:
I have the same issue. Did u manage to solve it?

palgy, 2023-07-27:
Yes, you need to make sure that you select right setting for the cloud
build