betaflight/betaflight-configurator #2855: 8j motor resource issue with BetaFPV f722 aio

Drone65, 2022-03-14:
### Describe the bug

When the led strip is selected in the configuration tab betaflight 4.3 stops seeing motor output in the motors. 

### To Reproduce

To reproduce this just turn led strip on and off and returning to the motors tab. 

### Expected behavior

I expected to be able to turn ledstrip on to program leds. But this issue is preventing that. 

### Configurator version

10.8 debug-ea6958f

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

haslinghuis, 2022-03-14:
What do you mean with `8j`? Please provide `diff` and `resource show all` between 3 backticks ``` on a separate line before and after the cli output.

Drone65, 2022-03-14:
Didn’t mean to leave the 8j on the email. And I’m new to all this. Do you mean a cli dump from the quad?