betaflight/betaflight-configurator #2859: usb_hid_cdc not working correctly

PeaceTW, 2022-03-18:
### Describe the bug

Setting usb_hid_cdc = on causes Betaflight not to detect FC anymore until FC is re-flashed. 
Using older FC firmware does not resolve issue.
Using Betaflight configurator 10.7.2 allows usb_hid_cdc to be enabled and also allows fc to be detected.

### To Reproduce

Windows 10
Betaflight configurator 10.8.0 RC3
FC is MATEKF722SE

To recreate---
1. Flash FC to 4.3.0 RC3
2. Go to CLI and type (set usb_hid_cdc = on)
3. save above command

After this I was not able to access FC configurator until FC was re-flashed in part 1.

### Expected behavior

FC should be able to be accessed but you cannot currently access them until re-flashed.

### Configurator version

10.8.0

### Flight controller configuration

```
# 
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Feb  1 2022 / 20:14:54 (3267f0417) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF722SE, version: b020e572, date: 2022-01-16T20:04:31Z

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id MTKS

# serial
serial 2 2048 115200 57600 0 115200

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ON_USB

# aux
aux 0 0 1 1425 2100 0 0
aux 1 1 0 900 1150 0 0
aux 2 13 3 1300 2100 0 0
aux 3 28 0 1825 2100 0 0
aux 4 35 2 1300 2100 0 0
aux 5 36 5 1300 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 1W2

# master
set gyro_lpf1_static_hz = 287
set gyro_lpf2_static_hz = 575
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 287
set gyro_lpf1_dyn_max_hz = 575
set acc_calibration = 62,42,266,1
set mid_rc = 1505
set min_check = 1005
set max_check = 1995
set dshot_idle_value = 400
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set small_angle = 180
set deadband = 7
set yaw_deadband = 7
set pid_process_denom = 4
set simplified_gyro_filter_multiplier = 115
set osd_tim_2_pos = 2516
set osd_mah_drawn_pos = 2083
set osd_altitude_pos = 2498
set osd_warnings_pos = 14793
set osd_avg_cell_voltage_pos = 2100
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 4
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5806
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 78
set dterm_lpf1_dyn_max_hz = 157
set dterm_lpf1_static_hz = 78
set dterm_lpf2_static_hz = 157
set p_pitch = 56
set i_pitch = 101
set d_pitch = 53
set f_pitch = 173
set p_roll = 51
set i_roll = 92
set d_roll = 46
set f_roll = 158
set p_yaw = 51
set i_yaw = 92
set f_yaw = 158
set d_min_roll = 34
set d_min_pitch = 39
set simplified_master_multiplier = 110
set simplified_d_gain = 105
set simplified_pi_gain = 105
set simplified_feedforward_gain = 120
set simplified_pitch_pi_gain = 105
set simplified_dterm_filter_multiplier = 105

rateprofile 0

# rateprofile 0
set thr_mid = 0
set thr_expo = 60
set rates_type = BETAFLIGHT
set roll_rc_rate = 165
set pitch_rc_rate = 165
set yaw_rc_rate = 150
set roll_expo = 22
set pitch_expo = 22
set yaw_expo = 20
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 65

# end the command batch
batch end
```

### Add any other context about the problem that you think might be relevant here

Note: currently don't have a second instant of BF on a different computer. Will attempt at a friends in the future soon.

I have left my diff setup below but don't know if it will help. This dump doesn't have the usb_hid_cdc set to on. This is caz if I do, well I can't access CLI. :(

haslinghuis, 2022-03-18:
On options tab please try enabling 

[image]

PeaceTW, 2022-03-18:
Yeah that did it! Thanks for the quick help.

closed 2022-03-18 by PeaceTW

thytum, 2022-04-23:
hello all :), i have the same issue but i cant find this option on betaflight configurator version 10.8, windows 11, i already did on cli set usb_hid_cdc = on, this thing of  show all serial is it in the configuration tab ? i have the expert mode enable, may you help me please?

haslinghuis, 2022-04-23:
Please update configurator to latest RC5 version or use https://github.com/betaflight/betaflight-configurator-nightlies/releases/tag/v20220423.99 

You can find it in `Options` tab before making a connection in the menu bar.

thytum, 2022-04-23:
Obrigado for the fast answer man! its working like a charm now, wish you a great weekend :)

dkjroot, 2022-05-15:
The solution doesn't work for me (see #2924 for my configuration details) - I get the same behaviour whether or not "show all serial devices" is selected:
My available devices in each of the possible combinations are:

Before enabling HID mode (i.e. USB_HID_CDC=OFF):
- With show all serial devices off and nothing connected:  Manual selection : "/dev/rfcomm0"
- With show all serial devices on and nothing connected: Manual selection : "/dev/rfcomm0"
- With show all serial devices off and quad connected: COM3 - Betaflight STM32F411 / Manual selection : "/dev/rfcomm0"
- With show all serial devices on and quad connected: COM3 - Betaflight STM32F411 / Manual selection : "/dev/rfcomm0"

After enabling USB_HID_CDC=ON:
- With show all serial devices off and nothing connected:  Manual selection : "/dev/rfcomm0"
- With show all serial devices on and nothing connected: Manual selection : "/dev/rfcomm0"
- With show all serial devices off and quad connected: Manual selection : "/dev/rfcomm0"
- With show all serial devices on and quad connected: Manual selection : "/dev/rfcomm0"

i.e. enabling show all serial devices doesn't fix the problem in my case.  

Thank you for all your efforts and this wonderful software we all enjoy so much :)

haslinghuis, 2022-05-15:
Use

[image]

dkjroot, 2022-05-16:
I have tried Full chip erase, no change I'm afraid.  If it's not affecting anyone else I guess it's just my drone. When I have some time I'll try earlier firmware, I think it was on 4.2.5 when it worked and I haven't tried going back that far yet.