betaflight/betaflight-configurator #2860: COM Port not reading

pmapeso, 2022-03-20:
### Describe the bug

When I plug my FC (Iflight succex-e F405) into my PC, BF config 10.8 RC3 will not detect my COM port and it will not read it at all. I tried manual selection and that didn't work. I've tried with 3 computers and have had the same issue.

FC works fine with 10.7.2 Config.

### To Reproduce

Install BF config 10.8 RC3
Open BF 10.8 RC3
Plug in IFRC_succex-e-f405(STM32F405)

### Expected behavior

The same port detection as the past with other configurators. Plug in, Config detects port, go on your way and connect and configure it.

### Configurator version

10.7.2

### Flight controller configuration

```
# 

# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.4 Oct 20 2020 / 08:18:45 (fbcaf8c50) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_SUCCEX_E_F4, version: c7fde51e, date: 2020-08-26T04:30:34Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFLIGHT_SUCCEX_E_F4
manufacturer_id IFRC
mcu_id 002300183539471038353530
signature 

# name: XL5 Mapeso

# feature
feature -LED_STRIP
feature -DISPLAY
feature AIRMODE

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -DISARM_REPEAT

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led
led 0 0,0::L:0
led 1 1,0::L:0
led 2 2,0::L:0
led 3 3,0::L:0

# aux
aux 0 0 1 1700 2100 0 0
aux 1 1 3 1375 1600 0 0
aux 2 13 0 1925 2100 0 0
aux 3 35 3 1900 2100 0 0
aux 4 36 2 1800 2000 0 0

# rxrange
rxrange 0 1100 1900
rxrange 1 1100 1900
rxrange 2 1100 1900
rxrange 3 1100 1900

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 14 23 27 29
vtxtable powerlabels 25 200 500 800

# rxfail
rxfail 0 h
rxfail 1 h
rxfail 3 h

# master
set gyro_lowpass2_hz = 375
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 750
set acc_calibration = -9,-38,126,1
set baro_hardware = NONE
set min_check = 1000
set max_check = 2000
set serialrx_provider = CRSF
set rx_min_usec = 875
set rx_max_usec = 2125
set blackbox_p_ratio = 64
set dshot_idle_value = 800
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 15
set vbat_max_cell_voltage = 440
set current_meter = NONE
set ibata_scale = 1
set ibata_offset = 0
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set osd_warn_batt_warning = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_link_quality_alarm = 50
set osd_vbat_pos = 2434
set osd_rssi_pos = 26
set osd_link_quality_pos = 2358
set osd_rssi_dbm_pos = 2072
set osd_tim_2_pos = 2453
set osd_flymode_pos = 2413
set osd_vtx_channel_pos = 2059
set osd_crosshairs_pos = 237
set osd_current_pos = 2529
set osd_mah_drawn_pos = 2208
set osd_craft_name_pos = 2445
set osd_altitude_pos = 139
set osd_warnings_pos = 14666
set osd_avg_cell_voltage_pos = 2498
set osd_disarmed_pos = 2347
set osd_esc_tmp_pos = 2049
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 4
set vtx_freq = 5658
set esc_sensor_halfduplex = ON
set usb_hid_cdc = ON
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_harmonics = 1
set name = XL5 Mapeso

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 98
set dyn_lpf_dterm_max_hz = 238
set dterm_lowpass2_hz = 210
set vbat_pid_gain = ON
set anti_gravity_gain = 5000
set iterm_relax_cutoff = 10
set yaw_lowpass_hz = 70
set p_pitch = 85
set i_pitch = 70
set d_pitch = 43
set p_roll = 68
set i_roll = 65
set d_roll = 38
set p_yaw = 105
set i_yaw = 115
set level_limit = 90
set d_min_roll = 20
set d_min_pitch = 22
set auto_profile_cell_count = 4

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 98
set dyn_lpf_dterm_max_hz = 238
set dterm_lowpass2_hz = 210
set vbat_pid_gain = ON
set anti_gravity_gain = 5000
set iterm_relax_cutoff = 10
set yaw_lowpass_hz = 70
set p_pitch = 85
set i_pitch = 70
set d_pitch = 43
set p_roll = 68
set i_roll = 65
set d_roll = 38
set p_yaw = 105
set i_yaw = 115
set d_min_roll = 20
set d_min_pitch = 22
set motor_output_limit = 62
set auto_profile_cell_count = 6

profile 2

# profile 2
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dterm_lowpass2_hz = 180
set vbat_pid_gain = ON
set anti_gravity_gain = 5000
set iterm_relax_cutoff = 10
set yaw_lowpass_hz = 70
set p_pitch = 80
set i_pitch = 65
set d_pitch = 43
set p_roll = 63
set i_roll = 60
set d_roll = 38
set p_yaw = 100
set i_yaw = 110
set level_limit = 90
set d_min_roll = 20
set d_min_pitch = 22
set motor_output_limit = 75
set auto_profile_cell_count = 4

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 99
set pitch_rc_rate = 99
set yaw_rc_rate = 99
set roll_expo = 5
set pitch_expo = 5
set yaw_expo = 5
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = SCALE
set throttle_limit_percent = 95

rateprofile 1

# rateprofile 1
set rates_type = KISS
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set throttle_limit_type = SCALE
set throttle_limit_percent = 95

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

pmapeso, 2022-03-20:
For the config version I am using 10.7.2, but the bug was with 10.8.0 RC3.

haslinghuis, 2022-03-20:
Please try disabling GPS.
Maybe first try enabling show all serial devices in options tab.

pmapeso, 2022-03-20:
Thank you, that worked.

closed 2022-03-20 by pmapeso

haslinghuis, 2022-03-20:
@pmapeso which one?

pmapeso, 2022-03-20:
Show all serial devices, sorry