betaflight/betaflight-configurator #2863: Bug or feature : Gyro filter slider always deactivated

mrRobot62, 2022-03-21:
### Describe the bug

[image]

Even I set unindependend or dependent switches to on. The gyro-slider is always deactivated.
Not sure if this is a feature based on some other function or a bug.

If not a bug, maybe user should be informed, why slider is deactivated.

Found in BF-Configurator (under Mac OS 10.15.7)
[image]

### To Reproduce

everytime

### Expected behavior

Slider should usable

### Configurator version

see screen shot 10.8.0 RC3

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

mrRobot62, 2022-03-21:
Hi @haslinghuis - long time ago .... :-)
Yeap will check this, my thought was to test on RC3 to focus on current bugs or other thins and to avoid with nightlies to struggle maybe on other issues.

However, will install tomorrow Nightly FW and Nightly BF-Config

mrRobot62, 2022-03-21:
[FuryF4OSD_diff_BF_Conf_10.8.0-RC3.txt](https://github.com/betaflight/betaflight-configurator/files/8319359/FuryF4OSD_diff_BF_Conf_10.8.0-RC3.txt)

haslinghuis, 2022-03-21:
Using
```
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
```
we get 

[image]

Just enabling any gyro filter will enable the slider.

haslinghuis, 2022-03-21:
Notice ON / OFF state here.

mrRobot62, 2022-03-22:
I tried all permutations with both ON/OFF switches - nothing helps. Currently I do not have enough time for debugging, will do it maybe later. I assume, there is an other hickup, which caused this issue

mrRobot62, 2022-03-22:
OMG 😱- 
But on perspektive of an user, maybe it would be helpful to inform user, if using Gyro-Multiplier, gyro Lowpass should be activated - with all consequences behind.

However - thanks for helping

closed 2022-03-22 by mrRobot62