betaflight/betaflight-configurator #2871: no bidirectional dshot after reboot enabled

DM4DS, 2022-03-25:
### Describe the bug

activating bidirectional dshot is not enabled after reboot on 4.3 RC3

after 5+ tries I was able to activate it....disabled it again and same problem. 10 trys and it´s still disabled.

The Poles count field is also empty after activating the "slider" in 4.3 before reboot - after reboot there is no poles field cause bidirectional dshot is disabled. In 4.2.11 the count is autofilled with "14" before reboot

reflashed 4.2.11 and it works first try

### To Reproduce

flash 4.3.0 RC3 as you see below to the FC. after apllying standards, try to activate bidirectional DShot.

FC reboots and bidirectional DShot is still not activated

# Betaflight / STM32F7X2 (S7X2) 4.3.0 Feb  1 2022 / 20:14:54 (3267f0417) MSP API: 1.44
# config: manufacturer_id: EXUA, board_name: EXF722DUAL, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

### Expected behavior

activate bidirectional DShot first try as in all older versions

### Configurator version

# Betaflight / STM32F7X2 (S7X2) 4.3.0 Feb  1 2022 / 20:14:54 (3267f0417) MSP API: 1.44 

### Flight controller configuration

```
diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Feb  1 2022 / 20:14:54 (3267f0417) MSP API: 1.44
# config: manufacturer_id: EXUA, board_name: EXF722DUAL, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name EXF722DUAL
manufacturer_id EXUA
mcu_id 001b00263238510d36353235
signature 

# name: 7

# feature
feature GPS
feature TELEMETRY

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 4 2 115200 9600 0 115200

# aux
aux 0 0 0 1950 2100 0 0
aux 1 2 3 1300 1700 0 0
aux 2 46 3 1900 2100 0 0
aux 3 13 1 1300 1700 0 0
aux 4 31 2 1300 1700 0 0
aux 5 35 1 1900 2100 0 0
aux 6 37 4 1300 2100 0 0

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -30,-11,-20,1
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set motor_output_reordering = 3,2,1,0,4,5,6,7
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = 180
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_descent_dist = 50
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set deadband = 10
set yaw_deadband = 10
set osd_current_pos = 2095
set osd_gps_speed_pos = 2060
set osd_gps_lon_pos = 2062
set osd_gps_lat_pos = 2065
set osd_gps_sats_pos = 2057
set osd_altitude_pos = 2052
set osd_avg_cell_voltage_pos = 2092
set gyro_2_align_yaw = 900
set name = 7

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

### Add any other context about the problem that you think might be relevant here

I only had the Diff ALL while I´m back on latest BF 4.2 configurator, in which everything works flawless, sorry

Asizon, 2022-03-26:
Fixed in the last dev build, please try it

Asizon, 2022-03-26:
https://github.com/betaflight/betaflight-configurator-nightlies/releases

closed 2022-03-26 by Asizon

DM4DS, 2022-03-28:
Reflashed to 4.3 RC3 with the 6day old Developer Version and it worked flawless.

Thanks
Sandor - DM4DS