betaflight/betaflight-configurator #2873: Can't enable DSHOT

Martin-TnT, 2022-03-26:
### Describe the bug

Device : EMAX Tiny Hawk II - with Betaflight 4.2.11 and Bluejay 0.16. 

Can't enable Bidirectional DShot with Betaflight Configuratior 10.8 RC3 - on another machine with Configuratior 10.7 there isn't any problem.

After Configuratior 10.8 RC 3 looks like DShot is disabled, but when enabling and setting the same parameters as in 10.7 the "Save and Reboot" is greyed.

And telematry is showing for each motor.

### To Reproduce

When enabling it in 10.8 RC3, the Motor poles is empty - enter 12 and save and reboot. 

When connection, DShot is deselected - no telemetry from the motors is showing.

First after enabling DShot is configurator 10.7, the telemetry is showing for each motor.

### Expected behavior

As in Configurator 10.7, when enabling  - Save and Reboot - and is is still enabled.

### Configurator version

10.8 RC 3

### Flight controller configuration

```
2022-03-26 @14:54:53 -- OS: Windows

2022-03-26 @14:54:53 -- Configurator: 10.8.0 (340087a4)

2022-03-26 @14:54:53 -- Using cached release information for configurator releases.

2022-03-26 @14:54:56 -- Serial port successfully opened with ID: 1

2022-03-26 @14:54:59 -- Serial port successfully closed

2022-03-26 @14:55:00 -- Serial port successfully opened with ID: 2

2022-03-26 @14:55:10 -- No configuration received within 10 seconds, communication failed

2022-03-26 @14:55:10 -- Serial port successfully closed

2022-03-26 @14:55:13 -- Serial port successfully opened with ID: 3

2022-03-26 @14:55:23 -- No configuration received within 10 seconds, communication failed

2022-03-26 @14:55:23 -- Serial port successfully closed

2022-03-26 @14:55:25 -- Serial port successfully opened with ID: 4

2022-03-26 @14:55:35 -- No configuration received within 10 seconds, communication failed

2022-03-26 @14:55:35 -- Serial port successfully closed

2022-03-26 @14:55:46 -- Serial port successfully opened with ID: 5

2022-03-26 @14:55:46 -- MultiWii API version: 1.43.0

2022-03-26 @14:55:46 -- Flight controller info, identifier: BTFL, version: 4.2.11

2022-03-26 @14:55:46 -- Running firmware released on: Nov 9 2021 20:28:23

2022-03-26 @14:55:46 -- Board: MTKS/MATEKF411RX(STM32F411), version: 0

2022-03-26 @14:55:46 -- Unique device ID: 0x6300593130511030343830

2022-03-26 @14:55:46 -- Craft name: TinyHawk II

2022-03-26 @14:55:46 -- Arming Disabled

2022-03-26 @14:55:49 -- CLI mode detected

2022-03-26 @14:57:24 -- Serial port successfully closed

2022-03-26 @14:57:26 -- Failed to open serial port

2022-03-26 @14:57:29 -- Serial port successfully opened with ID: 6

2022-03-26 @14:57:29 -- MultiWii API version: 1.43.0

2022-03-26 @14:57:29 -- Flight controller info, identifier: BTFL, version: 4.2.11

2022-03-26 @14:57:29 -- Running firmware released on: Nov 9 2021 20:28:23

2022-03-26 @14:57:29 -- Board: MTKS/MATEKF411RX(STM32F411), version: 0

2022-03-26 @14:57:29 -- Unique device ID: 0x6300593130511030343830

2022-03-26 @14:57:29 -- Craft name: TinyHawk II

2022-03-26 @14:57:29 -- Arming Disabled

2022-03-26 @15:06:21 -- CLI mode detected

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F411 (S411) 4.2.11 Nov  9 2021 / 20:28:23 (948ba6339) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF411RX, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name MATEKF411RX
manufacturer_id MTKS

# name: TinyHawk II

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# serial
serial 1 2048 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1300 1700 0 0
aux 4 35 2 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200

# master
set gyro_lowpass_hz = 0
set dyn_lpf_gyro_min_hz = 0
set acc_trim_pitch = -4
set acc_calibration = -92,29,41,1
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set rssi_offset = 20
set rx_spi_protocol = FRSKY_D
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 320
set vbat_warning_cell_voltage = 330
set small_angle = 180
set osd_warn_batt_not_full = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_cap_alarm = 450
set osd_vbat_pos = 2095
set osd_rssi_pos = 168
set osd_tim_2_pos = 2086
set osd_flymode_pos = 2081
set osd_vtx_channel_pos = 2452
set osd_current_pos = 2102
set osd_craft_name_pos = 73
set osd_display_name_pos = 394
set osd_pid_roll_pos = 353
set osd_pid_pitch_pos = 34
set osd_pidrate_profile_pos = 186
set osd_warnings_pos = 2433
set osd_avg_cell_voltage_pos = 374
set osd_disarmed_pos = 2403
set osd_flip_arrow_pos = 2422
set osd_rate_profile_name_pos = 23
set osd_pid_profile_name_pos = 2
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_total_time = ON
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 1
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5843
set vcd_video_system = NTSC
set frsky_spi_tx_id = 98,136
set frsky_spi_offset = -31
set frsky_spi_bind_hop_data = 0,30,60,91,120,150,180,210,5,35,65,95,125,155,185,215,10,40,70,100,130,160,190,221,15,45,75,105,135,165,195,225,20,50,80,110,140,170,200,230,25,55,85,115,145,175,205,0,0,0
set gyro_1_align_yaw = 1800
set name = TinyHawk II

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 0
set dyn_lpf_dterm_max_hz = 250
set dterm_lowpass_hz = 100
set dterm_lowpass2_hz = 200
set feedforward_transition = 25
set acc_limit_yaw = 100
set crash_recovery = ON
set iterm_relax = RPY
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set iterm_windup = 40
set iterm_limit = 150
set throttle_boost = 4
set p_pitch = 105
set i_pitch = 125
set d_pitch = 95
set f_pitch = 100
set p_roll = 105
set i_roll = 125
set d_roll = 95
set f_roll = 100
set p_yaw = 95
set i_yaw = 95
set d_yaw = 45
set angle_level_strength = 60
set horizon_transition = 80
set level_limit = 30
set horizon_tilt_effect = 80
set horizon_tilt_expert_mode = ON
set d_min_roll = 0
set d_min_pitch = 0
set auto_profile_cell_count = 1
set launch_control_mode = PITCHONLY

rateprofile 0

# rateprofile 0
set thr_expo = 50
set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 20
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60
set tpa_rate = 25
set tpa_breakpoint = 1650

# end the command batch
batch end

```

DM4DS, 2022-03-26:
Looks like the same problem I reported yesterday:

https://github.com/betaflight/betaflight-configurator/issues/2871

Asizon, 2022-03-26:
Fixed in development build please try it.

Asizon, 2022-03-26:
https://github.com/betaflight/betaflight-configurator-nightlies/releases

closed 2022-03-26 by Asizon

Martin-TnT, 2022-03-26:
Just tested 10.8.0-debug-99a1168 (link above) and yeah issue is fixed.

Thank for the update, and you hard work.

@DM4DS - weird that yours post didn't show up when I searched for similar issues... I properly misspelled or something...

Have  a nice weekend.