betaflight/betaflight-configurator #2874: "Ready" Mode OSD for Racing

bberg115, 2022-03-28:
### Is your feature request related to a problem? Please describe

Racing typically requires an individual or race director to verify that everyone is ready by asking for a thumbs up. In big events, this may require a 2nd person (race line) to coordinate. Although it is not critical, this requires additional volunteer/time and is not efficicent.

### Describe the solution you'd like

Use OSD to notify director that everyone is ready via the video feeds. This could be done a couple of ways but I think using "pre-arm" mode may be the easiest. When a pilot is in "pre-arm" mode, the OSD would show "READY" and may eliminate the need for "thumbs up" requests all together.

### Describe alternatives you've considered

This could also be done with a standalone mode/switch that the pilots could trigger.

bberg115, 2022-03-31:
> @bberg115 <https://github.com/bberg115> this looks good, maybe adding
> another switch for that. I dont think using prearm switch is the best
> option, because i know alotof users uses the non static switch in the
> transmitter, and then they should retain this switch

Good point.

 I can see this being a separate mode / option in the configurator so that
the user can decide if they want it on the same switch or use a switch for
more than 1 mode.

jonmahoney15, 2022-07-18:
So would this need to go away once flying or say something else like shred, burn, fly or rip?

bberg115, 2022-07-18:
My thought is that it would turn off after throttling up or by changing the
switch state to cancel. In order to re-initiate on the same pack, you would
need to change the switch again before it would show up.

jonmahoney15, 2022-07-18:
Would you want it to say anything when in a not ready state?

bberg115, 2022-07-18:
My initial thought is no, but that may be a nice feature.

There's some situations where it could be tricky to implement if the mode
isn't enabled by a standalone switch.

PR #3054 (2022-10-25): Ready Mode
Ready Mode allows the pilot to signal they are ready. Ideal for race lines so all pilots are ready to race.

Fixes: #2874
closed 2022-10-25 by haslinghuis