betaflight/betaflight-configurator #2877: Unable to flash Matek 405std

Smyffy82, 2022-03-30:
### Describe the bug

(Matek 405std
Opto 20amp ESCs 
RaceStar 2205 motors 
Matek vtx)
Unable to flash Matek 405std 
put flight controller into Dfu mode held the button on the board plugged into usb flashing gets halfway on screen then “closes usb successfully” freezes halfway doesn’t progress

### To Reproduce

Matek 405std 
Download target firmware 
Firmware flash 
Full chip erase 
Baud rate auto (changed baud rate to manual in case this was issue)

### Expected behavior

Firmware flashes successfully without any issues closes out of betaflight without issue.
Opening the application without issue seeing the flight controller working on screen 

### Configurator version

Latest version of BetaFlight

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

mrRobot62, 2022-03-30:
I assume you used BF-Configurator 10.8.0-RC3? And latest BF-FW (4.3-RC3) right?

Smyffy82, 2022-03-30:
Yes tryed this also 
Displays on top “ left USB device successfully closed” but doesn’t flash

Smyffy82, 2022-03-30:
[image]

BottomsUpFPV, 2022-03-30:
Experienced similar issue when flashing Matek H743 this week on Mac.
Attempted to flash RC2 from Matek's website, as well as RC3 and nightly from configurator.
Shows "USB device successfully opened" then immediately "USB device successfully closed" without flashing.
Configurator hangs with 'flash firmware' button lit, all others grayed out until I switch tabs then return.  Many hours of troubleshooting later... different cables, mac reboots, reinstalled configurator(10.8.0), learned that win7 is no longer supported, etc.
Issue resolved itself the following day when I was trying to reproduce in order to get screenshots.
Not sure what happened, or how I managed to fix the issue other than walking away for a few hours.

mrRobot62, 2022-03-31:
@BottomsUpFPV 
I'm a Mac-User too, are u working on Catalina or later (hopefully not earlier). Sometimes I have more or less same issues (with Catalina) with connecting FC via USB - most time (>95%) it works as designed. I'm using different FC-Hardware types, not sure if I can point this issue into a hardware specific or OS X  problem - not sure.
I have a Matek F405 (Hub & vTx) Stack, will check this next days

However, not sure how we can help you? 

@Smyffy82 
thems there is something wrong with your image upload - I can't see anything

BottomsUpFPV, 2022-03-31:
Track record had been 100% successful until this happened despite running Mojave still(I know, I know) across multiple FC.
Since updating to 10.8.0 , configurator has worked perfectly otherwise. Had wondered if it was related to the update. Can't reproduce it. Flashing properly now. Mac life?

graymattertech, 2022-03-31:
I'm having the same issue with my xilof4. 

Attempt to connect to PC but nothing shows in device manager. Put it in bootloader and it shows correctly STM BOOTLOADER. 

Attempt to flash with BF 10.7.2 and it gets half way  through and then says (USB device successfully closed), but it still shows in bootloader that it's connected to computer. I have exhausted all possible options to try and get this to work. 

Any thoughts or comments would be so helpful! Talk soon

haslinghuis, 2022-03-31:
Fixed in: https://github.com/betaflight/betaflight-configurator/pull/2836

graymattertech, 2022-03-31:
> Fixed in: #2836

Updated to 10.8.0 and trying to flash bf4.3 and still having issues. Click flash firmware and it now says fireware image contains addresses not found on target.

mrRobot62, 2022-03-31:
> > Fixed in: #2836
> 
> Updated to 10.8.0 and trying to flash bf4.3 and still having issues. Click flash firmware and it now says fireware image contains addresses not found on target.

mhm mhm - I made dozens of tests on my Mac with Catalina and with Betaflight 10.8.0-RC3 and with different nightly builds. Currently I do not have problems with DFU mode. Checked this with at least four different hardware types of controllers.

Unfortunately I do not have a Matek H743 nor a xiloF4 :-O - and maybe it's not absolutely impossible, that there is a problem with DFU, based on a hardware type, I try to analyze this....

closed 2022-05-09 by github-actions[bot]